luc-github/ESP3D #826: [BUG]Issue with ESP-01S / ESP8266 uploading files via ESP3D WEBUI

CelliMatti, 2022-10-24:
Dear all,
I have recently purchased the Bigtreetech SKR PRO v1.2 Mainboard, the TFT35-E3 v3.0 Display and the ESP-01S WIFI-Module. I was able to install them, but now I have following Problem:

Since I have the SD-Card (16gb) plugged into the mainboard slot, I am able to upload some printing files (named .gco) over the ESP3D WEBUI on the SD-Card. But unfortunately, I can only upload very small files, which means that as soon as I want to upload a file that has for example 3Mb it starts to upload but after a few minutes an error message appears (“Error 0“, see Picture).
[image]

The WIFI Module ESP-01S is attached to the WIFI pin-slot of the SKR PRO v1.2 mainboard and the SD-Card is also plugged into the mainboard slot. Furthermore, I have a TFT35-E3 v3.0 Touch Display attached to my printer.
I am running the mainboard and the WIFI-Module at a Baudrate of 250000 and set the SERIAL_PORT to 1 and the SERIAL_PORT_2 to 6 in Marlin.

I think that I found the problem, but I don´t know how to fix it:

I found out that when a file gets uploaded via the ESP3D WEBUI, the file gets saved on the onboard buffer of the WIFI-Module. As you can see in the picture my available size for SPIFFS about 228KB and whenever a file is bigger than this value, it can´t be uploaded.
In the other picture you can see that the buffer size is defined in the firmware of the ESP3D (MAX_RESEND_BUFFER), but I can´t increase this value.
[image]
[image]

In the firmware (esp firmware -> syncwebserver.cpp) you can also see that the available and the needed buffer size is checked every time.
So, the help that I need from you is following:
The files should not be buffered on the ESP-Module but should directly be transmitted to the SD-Card of the SKR PRO Mainboard. In this way the buffer size is not affected.
Or at least, the files should be transmitted in “Block Mode”, so that every file is divided up into smaller parts that could be buffered and transmitted to the SD card respectively.

Looking forward to your reply.

Kind Regards.

luc-github, 2022-10-24:
no that is not the issue
 FYI there is no file transfer protocol but GCODE transfer : M28 / M29,  because ESP board does not have direct access to printer SD card  
so your comment : 
>The files should not be buffered on the ESP-Module but should directly be transmitted to the SD-Card of the SKR PRO Mainboard. is not valid - if have direct access I would not bother to buffer files ^_^

This protocol M28/M29 is very sensitive and do not allow other commands to be used while in use, there was some update in BTT TFT to be sure the communication is stopped during upload but frequently the code is broken so it may be your issue

Additionally this protocol so slow like hell  (https://github.com/luc-github/ESP3D/discussions/535) that actually people use it once / complain about it, then stop to use it because it is useless for big files, so I stopped to support it.

The communication between ESP and Printer board should be stopped when upload is ongoing, Marlin does not stop sending report, and eventualy TFT continue to do polling which messup this fragile protocol

Usually `Error 0` means connection lost due to esp3d crash or unexpected behavior 

You may enable ESP3D debug to check what is wrong

CelliMatti, 2022-10-24:
Thank you for your explanation. 
If this protocol is so slow and sensitive, I don´t think that I am going to use it anymore, since it should be a well working system.
Do you know if there is a more powerful WIFI-Module or something more stable that I can use to control my printer an upload files (GCODE) via WIFI or Ethernet? 
(I don´t want to use a Raspberry Pi, but a system that could be connected directly to the SKR PRO mainboard.)

Thank you.

luc-github, 2022-10-24:
Did you read the link I gave you in above message ?

CelliMatti, 2022-10-24:
so, do you think I can use the WiFi back pack (the stand alone wifi module - https://www.panucatt.com/ProductDetails.asp?ProductCode=WB8266) and it will work properly?

luc-github, 2022-10-24:
the standalone module must replace an existing configured SD Card reader for Marlin, so SD pins must be connected to the ones used by Marlin

CelliMatti, 2022-10-24:
OK, thank you. 
Would you rather recommend the Toshiba flashair or the standalone module?

luc-github, 2022-10-24:
If you do not have toshiba flashair I do not think you will buy it (https://www.amazon.com/s?k=toshiba+flashair)- they are now over priced like crazy .... 
The standalone is just matter on how you connect to board.

luc-github, 2022-10-24:
the best solution would be : https://github.com/luc-github/ESP3D/issues/575
but I am not ready sorry

CelliMatti, 2022-10-24:
May I ask you one other thing ;) 
When I plug the SD-Card into the mainboard slot of the SKR PRO and I turn the printer on, the SD-Card is not recognized by the printer. So, I have to unplug it and plug it back in while the printer is turned on. Then it works.
Is there a line in Marlin when the card gets refreshed?

luc-github, 2022-10-24:
I do not know sorry, I guess there is a setting to mount on start

CelliMatti, 2022-10-24:
Thank you anyway for responding.

luc-github, 2022-10-24:
no problem Ibclose issue then

closed 2022-10-24 by luc-github