luc-github/ESP3D #835: Creality3D v1.1.3  connection available, but cannot print loaded files [BUG]

octavius08, 2022-11-18:
**Describe the bug**

Servus, I have an ESP32 dev with SPI bound SD card. After flashing and setting up ESP3D, the files can be uploaded, but after pressing the Play icon, the printer do nothing,

Printer: Creality3D v1.1.3 
ATMEGA1284P  8bit
Merlin bugfix-2.0.x 2019-07-05 
Bilinear Leveling
Extruder: 1
MightyCore/ATmega1284 setup:  Sanguino BOARD_MELZI_CREALITY

**To Reproduce**

1. Go to ESP Dashboard / Files
2. Click on "Play" on file exp.  AA.GCO (see the first screenshot)
3. The printer do nothing
4. No error, just in Terminal : 

M23 /AA.GCO
M24
ok
ok
ok T:18.91 /0.00 B:18.87 /0.00 @:0 B@:0

**Expected behavior**

the printer should move an start to print :)

**Screenshots**
[image]

[image]

[image]

**ESP3D Firmware:**

- ESP3D FW version: 3.0.0-a25.M2
- ESP3D-WebUI Version: 3.0.0.a218
- Wifi mode:AP
- Flash method: Arduino Ide 
- Any change done in FW:   just add Support for SD card:

#define SD_DEVICE_CONNECTION ESP_DIRECT_SD
#define SD_DEVICE ESP_SD_NATIVE

**Target Firmware:**

**M115**

FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0
ok

**M501**
echo:V67 stored settings retrieved (594 bytes; crc 37154)
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D1.75
echo:  M200 D0
echo:Steps per unit:
echo: M92 X80.00 Y80.00 Z400.00 E97.90
echo:Maximum feedrates (units/s):
echo:  M203 X500.00 Y500.00 Z12.00 E120.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X9000.00 Y9000.00 Z500.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R1500.00 T1000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
echo:  M205 B20000.00 S0.00 T0.00 X10.00 Y10.00 Z0.20 E2.50
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 W I0 J0 Z0.10750
echo:  G29 W I1 J0 Z0.04000
echo:  G29 W I2 J0 Z-0.03000
echo:  G29 W I0 J1 Z0.12750
echo:  G29 W I1 J1 Z0.18000
echo:  G29 W I2 J1 Z0.17000
echo:  G29 W I0 J2 Z0.27750
echo:  G29 W I1 J2 Z0.20000
echo:  G29 W I2 J2 Z0.81250
echo:Material heatup parameters:
echo:  M145 S0 H185 B45 F255
echo:  M145 S1 H240 B0 F255
echo:PID settings:
echo:  M301 P21.73 I1.54 D76.55
echo:Z-Probe Offset (mm):
echo:  M851 Z-1.87
ok

**Board used (please complete the following information):**

- ESP32 dev
- NodeMCU
- Flash size: 4M

**Additional context**
Also no data is displayed under UI/Settings/Merlin see picture above (second one).
The communication via TX/RX works: temperature, movement, fan can be controlled, it just doesn't print. i can still print from Octoprint and directly from the SD card on the printer. 
I've been working on this for 7 days now and I'm not making any progress.

luc-github, 2022-11-18:
well communication looks working so I do not think it is ESP3D issue.
What's happen if you enter manually in webui terminal the set of commands :
```
M21
M20
M23 /AA.GCO
M24
```

octavius08, 2022-11-19:
Many thanks for your response,
I connected the SD card like this:
[image]
I can list and upload files via ESP3D UI to the SD card.
I also tried version 2.1.2, but unfortunately the upload to SD doesn't work. However, Merlin data is displayed, control and monitoring of the printer works

octavius08, 2022-11-19:
Other attempt, I inserted the SD card with some files in to the slot on Ender3 board , the files were displayed as Extra SD and can be printed :) unfortunately uploading via WIFI is not possible, only refresh icon is displayed (see the picture)

[image]

The advantage of the SD card connected to the ESP32 via SPI is the upload speed, so I chose this connection beforehand.

luc-github, 2022-11-20:
the streaming from SD with esp3D is not yet available, it is planned but far being ready.

octavius08, 2022-11-20:
The streaming function from ESP32 board with an SD card would solve many upload and speed problems.
I recently bought a cheap CNC (GRBL Controler) milling machine, there is via cable an external control interface with an ESP32 and a card slot installed, the file upload seems to work. Unfortunately the firmware  is closed and not on github.

[image]

So we are eagerly awaiting the update, best regards.

luc-github, 2022-11-20:
for grbl the basic stream seems ok, it do same as the device you showed

octavius08, 2022-11-21:
Yes, that seems to be the firmware, but I also noticed that the CNC milling machine pauses with larger files to process the data, something like that is acceptable with a CNC but not with a 3D printer :)

luc-github, 2022-11-21:
Yes as I wrote for grbl the basic stream is ok because it is simple ping pong command/ack
For 3D printer it is more complex : https://github.com/luc-github/ESP3D/issues/243
That is why it is not yet implemented in esp3d - I need time to work on it

liqijian101, 2022-11-23:
https://oshwhub.com/liqijian101/esp3d_esp32__camer-v2

You can try this module. I have successfully tested. And the transmission speed is faster, about 500K/s

hazy1710, 2022-11-26:
Try "M23 /aa.gco" 
lowercase.
This helped me!
In the future, I replaced the lines in the web interface.
By adding the line "a=a.toLowerCase();"
`				a=a.toLowerCase();
				var i = {
					name: a,
					size: s,
					isdir: l,
					datetime: "",
					isprintable: files_showprintbutton(a, l)
				};`

hazy1710, 2022-11-26:
But I have another problem. When uploading files, the message "Reset Numbering failed" appears.

luc-github, 2022-11-26:
the lower case issue is Marlin issue - it should be fixed in Marlin or original repository you get the FW
If M20 list AA.GCO M23 should handle same name not another one

> "Reset Numbering failed" 
Means the `M110 N0` command failed (https://marlinfw.org/docs/gcode/M110.html)

luc-github, 2022-11-26:
The lower case issue looks similar to this old one: https://github.com/luc-github/ESP3D/issues/273

hazy1710, 2022-11-26:
It looks like it's worth updating the firmware in the printer. Thanks.

luc-github, 2022-11-28:
I think issue can be closed then

closed 2022-11-28 by luc-github