luc-github/ESP3D #849: [BUG] psram: PSRAM ID read error: 0xffffffff

atsju, 2023-01-14:
**Describe the bug**

I'm trying to flash [3.0.0-alpha3](https://github.com/luc-github/ESP3D/releases/tag/v3.0.0-alpha3) on [ESP32-cam board](https://fr.aliexpress.com/item/32981449079.html) and I always get error `psram: PSRAM ID read error: 0xffffffff`

I made some research  and I tested this sketch as you requested with different board configurations in arduino IDE : https://github.com/luc-github/ESP3D/issues/781#issuecomment-1166388898
Length is always 0, and there is always error `psram: PSRAM ID read error: 0xffffffff`

I have also seen in release note that `Remove No PSRAM camera`. 

Is this board supposed to have PSRAM ? 
Please advice, is the board I selected not usable anymore in this project because no PSRAM ? 

I know I have been able to make it work in 3.0 branch before the different releases and decided to wait for more stable version before using it. Can you advice if I need a different board or if something is wrong in my method (I tested on 3 boards) ?

luc-github, 2023-01-14:
the camera support in ESP3D need psram yes
your error may be hardware issue - for example the  ESP32 CAM from diymore seems having this issue on some boards https://amzn.eu/d/gyJ9tCG they do exchange to customer that has this issue 

the board your linked has 4MB of PSRAM but looks so cheap and seems a clone of ai-thinker camera which lead me to think it has same quality issue as  the one I mentioned

atsju, 2023-01-14:
OK I understand, thank you.

I have 3 boards from this seller, from 2 different batch/orders (and model differs a little between batches). I tried all 3 boards, and all have same error.

I'm also 100% sure the first board displayed a picture with working camera in 3.0 branch before the releases when I was trying to make it work early last year. So:
- Is there any time point where PSRAM was not needed in 3.0 branch ? (I understand it's needed now)
- Is there any possible (mis)configuration that I could try to change in code ?
I might remove shield and investigate to find root cause of the issue but I'm not convinced it's 100% hardware. Especially because it worked once. Not saying the boards are not shitty.

And as last resort:
- Can you recommend a known board to buy ?

luc-github, 2023-01-14:
well 3.0 and camera may work without psram but is is super slow
3.0 without camera enabled will work without psram

what is the output of 
'``
#include <Arduino.h>

void setup() {
  log_d("Total heap: %d", ESP.getHeapSize());
  log_d("Free heap: %d", ESP.getFreeHeap());
  log_d("Total PSRAM: %d", ESP.getPsramSize());
  log_d("Free PSRAM: %d", ESP.getFreePsram());
}

void loop() {}
```
and esp32-arduino 2.0.5 core? be sure of the version

do a test with and without camera enabled

atsju, 2023-01-15:
I had 2.0.3 in previous test. Now updated to 2.0.5 (note it's not latest)

### PSRAM enabled

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
E (118) psram: PSRAM ID read error: 0xffffffff
[     3][W][esp32-hal-psram.c:71] psramInit(): PSRAM init failed!
[    19][D][sketch_jan14a.ino:4] setup(): Total heap: 367920
[    19][D][sketch_jan14a.ino:5] setup(): Free heap: 344692 (or 339708 with board N1)
[    20][D][sketch_jan14a.ino:6] setup(): Total PSRAM: 0
[    24][D][sketch_jan14a.ino:7] setup(): Free PSRAM: 0
```
getsometimes `E (1039) spiram: SPI SRAM memory test fail. 131072/131072 writes failed, first @ 3F800000
`  every other reset
also if I do 2 resets in a row, there is a lockup and watchdog reset : `rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
`

### PSRAM disabled

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13864
load:0x40080400,len:3608
entry 0x400805f0
[    18][D][sketch_jan14a.ino:4] setup(): Total heap: 373244
[    19][D][sketch_jan14a.ino:5] setup(): Free heap: 350016
[    19][D][sketch_jan14a.ino:6] setup(): Total PSRAM: 0
[    23][D][sketch_jan14a.ino:7] setup(): Free PSRAM: 0
```

luc-github, 2023-01-15:
so as minimal script show the issue it is definitly not configuration issue - it is hardware issue for me sorry 

I suggest you to order genuine ai-thinker esp32 cam not cheap clone

atsju, 2023-01-15:
This was the plan, but I think "genuine AI-thinker" are not produced anymore. Maybe the copy sold by https://www.olimex.com/ (re-sold by mouser) is OK but they are out of stock.

Can you recommend a seller ?

luc-github, 2023-01-15:
I suggest to try your amazon reseller first for a replacement -  they should be aware of the issue 

well on Aliexpress I just typed esp32cam ai-thinker and gor several results like:
https://fr.aliexpress.com/item/1005002510653051.html

or you can rely on user feedback for clones

atsju, 2023-01-15:
The one you suggest has no review to rely on. And TBH I'm pretty sure it's also a clone/copy.

I'm really interested to know what the underlying HW issue is ...
I have seen shitty boards, and there is always a reason behind the problem.

luc-github, 2023-01-15:
check what chip they use for psram, it should be the root cause, external psram  are connected to pins pin 16/17

luc-github, 2023-01-15:
and check this one https://fr.aliexpress.com/item/1005001468076374.html

atsju, 2023-01-15:
they all use [this](https://www.mouser.fr/ProductDetail/AP-Memory/APS6404L-3SQR-SN?qs=IS%252B4QmGtzzqsn3S5xo%2FEEg%3D%3D) chip which seems to be compliant part number from my research (even if not the official [esp-psram64](https://www.espressif.com/sites/default/files/documentation/esp-psram64_esp-psram64h_datasheet_en.pdf)) I will try to see if I find the official chip for testing

Thank you for the links

atsju, 2023-01-15:
I bought Amazon one without too much of expectation. let's wait and see

atsju, 2023-01-15:
Additional note : esp-eye is an official dev-board from espressif. Maybe this should be the recommended camera module to use to have no issues at all and not depend on Chinese designs/clones.
Have you tried this board ?

luc-github, 2023-01-15:
sure you buy original espressif dev board it should work - it is just not same budget ^_^

No I did not used this board - I just put as the ones supported upstream - too expensive for me just for testing

atsju, 2023-01-17:
Got new boards with motherboard. And it seems **motherboard** was cause of issue. I did cross the boards for tests.

MB with one button is bas, MB with 2 buttons is good. I don"t know if it's linked to type of MB or if I fried it or other. But PSRAM was not faulty. Maybe this can help other people.

Issue can be closed. I will leave a message here if I find the exact issue on MB

closed 2023-01-17 by atsju