luc-github/ESP3D #850: [BUG]<Enter comprehensive title>

KingBain, 2023-01-15:
Just for others, 

was getting an error when building ESP3D-3.0.0-alpha3 in vscode, for an old chip :)  esp8266_1MB_wifi

```
Compiling .pioenvs\esp8266_1MB_wifi\liba60\WebSockets\libb64\cdecode.c.o
In file included from .piolibdeps\esp8266_1MB_wifi\WebSockets\src\libb64\cdecode.c:9:
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: data definition has no type or storage class
  132 | bool __attribute__((const, nothrow)) esp_is_8285();
      | ^~~~
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: type defaults to 'int' in declaration of 'bool' [-Wimplicit-int]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: 'const' attribute ignored [-Wattributes]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: 'nothrow' attribute ignored [-Wattributes]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:38: error: expected ',' or ';' before 'esp_is_8285'
Compiling .pioenvs\esp8266_1MB_wifi\liba60\WebSockets\libb64\cencode.c.o
  132 | bool __attribute__((const, nothrow)) esp_is_8285();
      |                                      ^~~~~~~~~~~
*** [.pioenvs\esp8266_1MB_wifi\liba60\WebSockets\libb64\cdecode.c.o] Error 1
In file included from .piolibdeps\esp8266_1MB_wifi\WebSockets\src\libb64\cencode.c:9:
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: data definition has no type or storage class
  132 | bool __attribute__((const, nothrow)) esp_is_8285();
      | ^~~~
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: type defaults to 'int' in declaration of 'bool' [-Wimplicit-int]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: 'const' attribute ignored [-Wattributes]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:1: warning: 'nothrow' attribute ignored [-Wattributes]
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:132:38: error: expected ',' or ';' before 'esp_is_8285'
  132 | bool __attribute__((const, nothrow)) esp_is_8285();
      |                                      ^~~~~~~~~~~
*** [.pioenvs\esp8266_1MB_wifi\liba60\WebSockets\libb64\cencode.c.o] Error 1
.piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp: In member function 'void WebSocketsServer::handleNewClients()':
.piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp:646:96: warning: 'WiFiClient WiFiServer::available(uint8_t*)' is deprecated: Renamed to accept(). [-Wdeprecated-declarations]
  646 |         WEBSOCKETS_NETWORK_CLASS * tcpClient = new WEBSOCKETS_NETWORK_CLASS(_server->available());
      |                                                                                                ^
In file included from C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:40,
                 from .piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSockets.h:156,
                 from .piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp:25:
C:\Users\Bain\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiServer.h:85:14: note: declared here
   85 |   WiFiClient available(uint8_t* status = NULL) __attribute__((deprecated("Renamed to accept().")));
      |              ^~~~~~~~~
.piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp: In member function 'WSclient_t* WebSocketsServerCore::handleNewClient(WiFiClient*)':
.piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp:634:12: warning: function may return address of local variable [-Wreturn-local-addr]
  634 |     return client;
      |            ^~~~~~
.piolibdeps\esp8266_1MB_wifi\WebSockets\src\WebSocketsServer.cpp:626:20: note: declared here
  626 |         WSclient_t dummy = WSclient_t();
      |                    ^~~~~
```

did some googling and this resolved the issue. Seems to be an upstream issue 
https://github.com/Links2004/arduinoWebSockets/issues/782#issuecomment-1380716111

luc-github, 2023-01-15:
I provide the libraries to use in libraries directory  - you should not have issue with them

luc-github, 2023-01-15:
I have fixed the issue to freeze esp8266 pio to 4.0.1 - thank you for the report 

configurator is also updated accordingly 

thank you for the report

closed 2023-01-15 by luc-github