luc-github/ESP3D #851: [FEATURE REQUEST]USB serial support on  ESP32-S2/ESP32-S3 for systems using  CP210x, FTDI FT23x or CP34x

luc-github, 2023-01-20:
Idf 5.1 has recently implemented multiple drivers registration support for USB CDC-ACM Virtual Com , it is implemented in [ESP3D-TFT project](https://github.com/luc-github/ESP3D-TFT/discussions/1#discussioncomment-4712100)

I have tested the feature with ESP3D-TFT with several boards (end3/ ender5/ FYSTEC E4, ESP32 boards...) and so far so good, but the esp board still need external power supply because USB port do not power esp,  but no need to find serial port connections on printer/cnc anymore, just use right cable to connect printer/cnc USB port to ESP32-S2/S3 and they can communicate, this long time request finally come true ^_^  

Arduino core dev team is working on IDF 5.1 implementation - https://github.com/espressif/arduino-esp32/blob/esp-idf-v5.1/
Once merged / released I will port ESP3D to this version and implement the USB-OTG support on ESP32-S2 and ESP32-S3.

liqijian101, 2023-01-20:
It is wonderful！！Can esp3d also have this function?

khayamgondal, 2024-02-11:
@luc-github any update on this? Did you merge in ?

luc-github, 2024-02-11:
no, core 3.0 is.not released

luc-github, 2024-02-13:
Working on porting USB-OTG from ESP3D-TFT to a ESP3D I have realized that IDF 5.1.2 release updated the USB drivers
before:
```
## IDF Component Manager Manifest File
dependencies:
  usb_host_ch34x_vcp: "^1"
  usb_host_cp210x_vcp: "^1"
  usb_host_ftdi_vcp: "^1"
  usb_host_vcp: "^1"
  idf: ">=5.1.0"
```

to 
```
## IDF Component Manager Manifest File
dependencies:
  usb_host_ch34x_vcp: "^2"
  usb_host_cp210x_vcp: "^2"
  usb_host_ftdi_vcp: "^2"
  usb_host_vcp: "^1"
  idf: ">=5.1.0"
```
which bring some minor API change  but also a RX buffer  with configurable size

[image]

So I will update the code to handle driver update in ESP3D-TFT first and review the porting code I have already did which consist in an additional arduino library using idf usb-otg component

luc-github, 2024-02-16:
First step is now done, I have created an arduino library https://github.com/luc-github/esp32-usb-serial
I need now to prepare esp3d code to support 5.1 IDF code from esp32-arduino 3.0.0-alpha 3, I  won't test platformio

not-jan, 2024-06-17:
Hi, esp32-arduino has been releases. Are there any updates on this? Can I help in implementing somewhere?

luc-github, 2024-06-17:
this feature was not  in release so I did a library https://github.com/luc-github/esp32-usb-serial

I have ported also the ssdp library because arduino 3.0  is a breaking change
but platformio seems having issue to support the arduino esp32 3.0
so I have put on hold the porting of esp3d 3.0 for the moment

luc-github, 2024-10-13:
Working slowly on implementation due to busy life, I moved from PIO to community PIO to get support of esp32 3.0.4~5
the library is integrated and the new client is almost implemented https://github.com/luc-github/ESP3D/tree/usb-otg
so it is WIP

luc-github, 2024-10-19:
done with https://github.com/luc-github/ESP3D/commit/b40937122a4b41740897f13530dc9625f5612805

closed 2024-10-19 by luc-github

luc-github, 2024-10-19:
Note: Configurator need to be updated to add 
```
/* Optional Output 
 * Instead of Serial, you can use USB Serial
 * USE_USB_SERIAL //for ESP32  S2/S3 Only
 */
#define USB_SERIAL_FEATURE
```