luc-github/ESP3D #853: Internet access error

Stephano2000, 2023-01-30:
Hi,
In order to provide remote internet access to the ESP3D UI, i forwarded ports 80 and 81 to a public server server:10554 and server:10555 respectively. However the UI keeps disconnecting with a websocket error after 1  or 2 seconds. Accessing ESP3D locally works fine.
I am not sure whether this is a bug or i am just not doing the correct port forwarding, noting that ports 80 and 81 are not available on the public server.
- ESP3D v3
- ESP32-CAM

luc-github, 2023-01-30:
yes this is due to a workaround for apple captive portal, it use local ip for websocket instead of `document.location.host` provided in address bar
It is an issue I forget to handle  orz... I need to think about it

luc-github, 2023-01-30:
well actually I use `ws://${document.location.hostname}:${connectionSettings.current.WebSocketPort}${path}` so my assumption is wrong it was already fixed

can you check the dev tool of your browser to see what url is used and what is the error ?
it could be due to antivirus / antispam software as you are no more on local area

Stephano2000, 2023-01-30:
It seems it's still after port 81:
"Firefox can’t establish a connection to the server at ws://45.33.36.41:81/."

luc-github, 2023-01-30:
ok so check what could block on your computer, disable antivirus , firewall to check which is the blocking point

Stephano2000, 2023-01-30:
Antivirus and firewall are not in the picture.
Why would the web browser try to open the websocket at port 81 while the base port is 10554?

luc-github, 2023-01-30:
webui websocket use 81

luc-github, 2023-01-30:
I did not handled you change websocket port, for nat

luc-github, 2023-01-30:
I will push a solution later tonight, I am out now

Stephano2000, 2023-01-30:
I changed the HTTP protocol port to 10554 in the UI, forwarded ports 10554 and 10554 to 10554 and 10555 respectively, and there are no disconnections anymore.
So the browser itself should set the websocket port to the HTTP host port + 1 so that the websocket connection is successful.

luc-github, 2023-01-30:
yes because you use same as used by FW
you can have a look by just typing [ESP800] in web ui terminal, it will show the webui websocket port

luc-github, 2023-01-30:
yes as I wrote I did not handled the case user change the port by nat - it currently use the port provided by [ESP800] - I will fix it in web UI by collecting current http port and add +1 to it to make the webui websocket port

luc-github, 2023-01-30:
can you test this one and feedback ?
[index.html.gz](https://github.com/luc-github/ESP3D/files/10536342/index.html.gz)

Stephano2000, 2023-01-30:
It bricked the UI. The UI flashed "connecting" for a fraction of a second then shows a continuously spinning arc.
What was the force upload url (in the hope that it works to avoid a reflash)?

luc-github, 2023-01-30:
arg you tested in local with port 80 not remote, I missed this case orz 
I fixed in this one 
[index.html.gz](https://github.com/luc-github/ESP3D/files/10536653/index.html.gz)
Sorry ...

luc-github, 2023-01-30:
use `http://your ip?forcefallback=yes` to update because ui is broken

Stephano2000, 2023-01-30:
> arg you tested in local with port 80 not remote, I missed this case orz I fixed in this one [index.html.gz](https://github.com/luc-github/ESP3D/files/10536653/index.html.gz) Sorry ...

I can confirm that this one doesn't disconnect both locally, and remotely with forwarded ports.
Thanks

luc-github, 2023-01-30:
Cool, thank you for the feedback - I need to update the test server and do a rebuild for all target and update some docs
It will be done tomorrow -  here it is almost 23h and doing now is recipe to trouble / mistake, I did enough for tonight ;)

liqijian101, 2023-01-31:
Can HTTPS be supported

luc-github, 2023-01-31:
current esp32 arduino webserver does not have this feature, it is planned as far I know

liqijian101, 2023-01-31:
The webpage displays "Webdevsocket connection error"
- [ ] Enable websocket
- [ ] Port 8282

luc-github, 2023-01-31:
@liqijian101 as issue here is confirmed resolved  by OP- your issue is different - so please open new issue for your problem with full reproduction steps

luc-github, 2023-02-01:
@Stephano2000  fixed in latest webui (a29), thank you for raising it

closed 2023-02-01 by luc-github