luc-github/ESP3D #895: Error: 'struct camera_config_t' has no member named 'pin_sccb_sda'

3D-Darryl, 2023-02-28:
When compiling 3.0 with Ai Thinker camera,  the above error pops up for pin_sccb_sda and pin_sccb_scl.  Both errors go away if the suggested fix  is done in camera.cpp  i.e. pin_sccb_sda -> pin_sscb_sda  and pin_sccb_sc -> pin_sscb_scl.
Not sure if this is correct fix as camera only acquires an image on powerup and never again.

luc-github, 2023-02-28:
what esp32 core version are you using ?

3D-Darryl, 2023-03-01:
I do not know how to find a code revision, but I downloaded it on 26-Feb.  The only commits I saw had to do with documentation.
I Tried a refresh of 100mS and 500mS, but never had any updates.

luc-github, 2023-03-01:
You use arduino check board manager and check the esp32 board

3D-Darryl, 2023-03-01:
The board is Ai-Thinker ESP32-CAM, what you meant by "core"?

luc-github, 2023-03-01:
https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-board-manager

3D-Darryl, 2023-03-01:
Thank you for your patience.  ESP32 Arduino 2.0.3 is installed

luc-github, 2023-03-01:
should be 2.0.6
Ide 2.X
[image]

Ide 1.8.X
[image]

3D-Darryl, 2023-03-09:
I want to thank you for your interest and support.  I struggled for 3 weeks
to get the esp8266  working, where my hope was to add a serial camera (but
there is probably not enough IO).  Then I found the ESP32 with a camera and
the promise of software to support it, but after struggling with that for 2
1/2 weeks i am left with an Arduino IDE that has become recalcitrant.  You
had suggested that allowing the update was my error, but I can't guarantee
that the Arduino or its libraries will never be updated, and in fact will
need updates for future projects.  So I cannot compile the camera code and
am left with no way forward.
[image: image.png]
I decided to stop banging my head against the wall and I put the project
aside to move to something more fruitful and less frustrating.  I plan to
have another look at it in April by which time perhaps wiser heads will
have found a solution.
Thank you again for your support.

luc-github, 2023-03-09:
I suggest you to use platformio instead of arduino ide - you won't have to handle anything because I already did :
http://esp3d.io/esp3d/v3.x/installation/index.html#setup-your-development-environment

luc-github, 2023-03-09:
I close issue then

closed 2023-03-09 by luc-github

3D-Darryl, 2023-03-09:
Luc,
    Platform IO has even more issues.  It no longer builds the Nordic
examples that worked last week, and it won't build the Marlin code anymore
because of (apparently) an issue with the crypto code.  I have not found a
solution there either.  But it never built the ESP32 code.
Darryl

luc-github, 2023-03-09:
well issue is your side yesterday the build was passing using arduino and platformio
https://github.com/luc-github/ESP3D/actions/workflows/build-ci.yml

3D-Darryl, 2023-03-09:
OK, thanks.

luc-github, 2023-03-09:
the auto build do fresh setup for each build so you may have a look to see what it is doing compare to your setup
https://github.com/luc-github/ESP3D/blob/3.0/.github/workflows/build-ci.yml

https://github.com/luc-github/ESP3D/tree/3.0/.github/ci