luc-github/ESP3D #897: ESP32 LoRa Heltec v2 with display oled 128x64- Display not working display

PatricLeal, 2023-03-01:
When trying to upload ESP3D to the Heltec Lora board, everything works perfectly, but the OLED display does not show any visual information.

As I tried to force declaration of pins in configuration.h but I was not successful in displaying the screen.

Configuration.h file section:
/***************************************************
*
* display settings
*
* rendering screens
*
************************************/

// printer screen
// Your printer has a display
#define PRINTER_HAS_DISPLAY

// ESP3D screen
// OLED I2C SSD1306 128X64
// Screen connected to ESP board
#define DISPLAY_DEVICE OLED_I2C_SSD1306_128X64

// Display i2C address
// Wire address of display
#define DISPLAY_I2C_ADDR 0x3c
#define ESP_SDA_PIN 0
#define ESP_SCL_PIN 15
#define ESP_RST_PIN 16
/***************************************************
*
* Levels of security
*
* How commands are allowed to be sent to ESP3D
*
************************************/

// Enable serial commands
// Allow commands to be sent to ESP3D via serial port
#define SERIAL_COMMAND_FEATURE
-------------
plataformio.ini file:

[platform]
src_dir = esp3d
build_dir = .pioenvs
lib_dir = libraries
libdeps_dir = .piolibdeps
data_dir = esp3d/data
default_envs = esp32_8MB_disp_wifi

[env:esp32_8MB_disp_wifi]
platform = espressif32@6.0.1
board = esp32dev
framework = arduino
monitor_speed = 115200
monitor_echo = yes
monitor_filters = send_on_enter, colorize, esp32_exception_decoder
; set frequency to 240MHz
;board_build.f_cpu = 240000000L
; set frequency to 80MHz
;board_build.f_flash = 80000000L
;board_build.flash_mode = qio
upload_speed = 460800
extra_scripts = pre:platformIO/extra_script.py
board_upload.flash_size = 8MB
board_build.partitions = default_8MB.csv
build_flags = -DCORE_DEBUG_LEVEL=0

[image]
[image]
[WiFi Lora32Manual.pdf](https://github.com/luc-github/ESP3D/files/10863967/WiFi.Lora32Manual.pdf)

*** In test applications the display of esp32 lora wifi works normally using SSD1306 by Heltec-Aaron-Lee library for display.

luc-github, 2023-03-01:
What Firmware version are you using ?

This board need a special setup on pin 16  to reset screen at start if I remember well

luc-github, 2023-03-03:
did it fixed your issue? it works on my board

luc-github, 2023-03-07:
I close issue if no feedback

closed 2023-03-07 by luc-github