luc-github/ESP3D #907: [BUG] Can't get configuration to show

mikedattolo, 2023-03-27:
*Can send commands to 3d printer but can't see the printer config at all

*Can send commands to 3d printer but can't see the printer config at all
set BTT octopus serial port to 3
flashed firmware to esp8266
connected to esp3d 
can send commands
cant see temps, files or printer configuration

**Expected behavior**
To see all information of printer 

To see all information of printer about my printer

**Screenshots**

**ESP3D Firmware:**

- ESP3D FW version: [current version latest release]
- ESP3D-WebUI Version:[e.g. 2.1b14] 
- Wifi mode:[2a4n-esp8266]
- Flash method: [visual stuio/ PlatformIO]  

**Target Firmware:**

- Name: [e.g. Marlin]
- Version [e.g. 2.0]

**Board used (please complete the following information):**
BTT Octopuss 

- MCU:[e.g. ESP32]
- Name:[e.g. NodeMCU2s]
- Flash size: [e.g. 4M: 2M/2M]

**Additional context**

mikedattolo, 2023-03-28:
anyone help plzzz

luc-github, 2023-03-29:
Do you use the ESP module supposed to be used with octopus on wifi socket ?

mikedattolo, 2023-03-29:
> Do you use the ESP module supposed to be used with octopus on wifi socket ?

No unfortunately, I couldn't obtain one

luc-github, 2023-03-29:
so what esp board are you using and how did you wire it ? because it is likely the issue

mikedattolo, 2023-03-29:
I am using this esp board: https://www.amazon.com/HiLetgo-Internet-Development-Wireless-Micropython/dp/B010O1G1ES/ref=asc_df_B010O1G1ES/?tag=hyprod-20&linkCode=df0&hvadid=309818716690&hvpos=&hvnetw=g&hvrand=4914738068086214845&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9002239&hvtargid=pla-361876718784&th=1
and it wired by using the BTT octopus wifi module port, using 3.3v Ground, and the TX and RX pins 
if needed here is the pin layout for the BTT octopus

mikedattolo, 2023-03-29:
https://teamgloomy.github.io/btt_octopus_pro_1.0_f429_pins.html

mikedattolo, 2023-03-29:
if you cant find the wifi module on the board look for section U10

luc-github, 2023-03-29:
What is [ESP420] output in Webui ?
What is the M115 output in WEBUI?

mikedattolo, 2023-03-29:
> What is [ESP420] output in Webui ?
> What is the M115 output in WEBUI?

What is ESP420? , and I can’t currently tell you the output at the moment for M115 because the printer is at my work shop and I am currently home

luc-github, 2023-03-29:
is an ESP command

luc-github, 2023-03-29:
and it is `[ESP420]`

luc-github, 2023-03-29:
then do another test when you will try
directly connect your nodemcu to your PC and use a serial terminal to communicate with the board - the one provided in platformio - select monitor or if you use arduino ide use the embedded serial monitor
and check what is the output of `[ESP420]` also

mikedattolo, 2023-03-29:
Just a tad bit confused, so when I go in tomorrow you want me to plug in the board to my computer and using platformio I select monitor and wait for a result?

luc-github, 2023-03-29:
yes select Monitor  wait for ESP to boot - then type `[ESP420]` + enter and check what is the output

mikedattolo, 2023-03-29:
> yes select Monitor wait for ESP to boot - then type `[ESP420]` + enter and check what is the output

And should I leave it connected to the 3-D printer when doing this?

luc-github, 2023-03-29:
no you take only the ESP
you can also open webui  and in serial terminal type `Hello` and verify the webui display `Hello`

luc-github, 2023-03-29:
this is to check your ESP board is working properly

mikedattolo, 2023-03-29:
> this is to check your ESP board is working properly

Gotcha OK, my only problem now is I have to wait until tomorrow to try and do this like I said before my printer is all the way at my work shop, sorry for the inconvenience but I will keep you updated with the results tomorrow

mikedattolo, 2023-03-29:
> this is to check your ESP board is working properly

so it appears I have no clue what I am doing lol, on the visual studio on platform io for the esp I hit monitor and I get 
Warning! skipping unknown filters esp8266_exception_decoder known filters are: colorize, debug, default, direct and etc
Termial on com4 | 250000 8-N-1
more details at: https//bit.ly/pio-monitor-filters
quit ctrl + c | menu ctrl + t | help ctrl+h
and when putting hello in the terminal on webui it just shows in commands hello no other input or output
and I can't type anything when doing monitor it all becomes red when doing so

luc-github, 2023-03-29:
I do not understand: webui become red ?
Please do screenshot

luc-github, 2023-03-29:
also the initial test was also :

>What is [ESP420] output in Webui ?
>What is the M115 output in WEBUI?

mikedattolo, 2023-03-29:
> also the initial test was also with esp connected to printer :
> 
> > What is [ESP420] output in Webui ?
> > What is the M115 output in WEBUI?
[image]
[image]

mikedattolo, 2023-03-29:
[image]

> What is [ESP420] output in Webui ? What is the M115 output in WEBUI?

M115 doesn't do anything 
but [esp420] gave this output:
{"chip_id":"1986173","cpu":"160","freemem":"33.37 KB","SDK":"2.2.1(cfd48f3)","flash_size":"4.00 MB","update_size":"534.11 KB","spiffs_size":"1.86 MB","baud_rate":"9600","sleep_mode":"Modem","channel":"1","phy_mode":"11g","web_port":"80","data_port":"8888","active_mode":"AP (2E:F4:32:1E:4E:7D)","ap_ssid":"MD5DPrinter","ssid_visible":"Yes","ssid_authentication":"WPA2","ssid_max_connections":"4","ssid_dhcp":"Started","ip":"192.168.0.1","gw":"192.168.0.1","msk":"255.255.255.0","connected_clients":[{"bssid":"F8:E4:E3:02:35:9B","ip":"192.168.0.100"}],"disabled_mode":"STA (2C:F4:32:1E:4E:7D) is disabled","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Notifications":"Disabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.3b0 ESP8266/8586"}

luc-github, 2023-03-29:
baud rate 9600??? Marlin is 250000 by default or 115200 if modified

luc-github, 2023-03-29:
that is normal your PIO serial monitor does not communicate it is set to 250000 
Baudrate must be same on both side to communicate

mikedattolo, 2023-03-29:
> baud rate 9600??? Marlin is 250000 by default or 115200 if modified

yeah but that's the thing I don't understand is the actual esp8226 says to use a bud rate of 9600 and when in the web UI with a rate at 250000 i can't send any commands to the printer whereas at 9600 I can

mikedattolo, 2023-03-29:
for the ports in marlin I have them set to 9600, but im guessing im gonna have to switch that now to 250000, and ill do the same for the esp3d and then ill check in the platform io

mikedattolo, 2023-03-29:
[image]
so this is new, the name of the esp and a command M117 and Ip show now, however typing hello in the platformio does nothing but typing in the webui shows in the platformio terminal

mikedattolo, 2023-03-29:
what does this mean?

luc-github, 2023-03-29:
you  mean typing `hi` + enter in PIO does not show anything in webui ?

luc-github, 2023-03-29:
Ok I am checking using your settings

mikedattolo, 2023-03-29:
so I can now control the 3d printer with a bud rate of 250000 now but I still can not pull up the printer config in the webui

luc-github, 2023-03-29:
I have tested with same board as you connected to my PC:
[image]
And no issue 
so if your ESP board connected to your PC only can send command  from web ui to Serial but not from Serial to Webui  it means your ESP board has issue

mikedattolo, 2023-03-29:
> I have tested with same board as you connected to my PC: [image] [ [image] ](https://user-images.githubusercontent.com/8822552/228560204-a3c81fa9-df69-4031-94bb-0a32a84efd7f.gif) [ [...]

I can send commands from the web UI and they show in the terminal like you just did there, my issue is I cant get the printer config to show

mikedattolo, 2023-03-29:
or oh wait hold on ok i see , yeah I cant send from terminal to webui, but then how do I fix that issue

luc-github, 2023-03-29:
your board has defect - I can do nothing - change the ESP board

luc-github, 2023-03-29:
The issue is pure hardware

mikedattolo, 2023-03-29:
> your board has defect, RX pin is not functional - I can do nothing - change the ESP board,

wait nope I am just stupid, I did monitor again, and with the webui OPEN!! this time lol XD I sent "hello" in the terminal and it appears in the webUI, Like I said before completely new to this

mikedattolo, 2023-03-29:
so that means the RX and TX pins are working, but then why cant I get the WebUI to pull up the printer config?

luc-github, 2023-03-29:
Check your Marlin configuration
M503 should give you the report

mikedattolo, 2023-03-29:
how do i do that?

mikedattolo, 2023-03-29:
like what am I looking for or need to do?

mikedattolo, 2023-03-29:
[image]
[image]
[image]
cause these are the only things I could find having to do with M503 and wifi

luc-github, 2023-03-29:
M503 is not related to wifi, I suggest you to read Marlin documentation, to see what need to enable M503 code

mikedattolo, 2023-03-29:
> Check your Marlin configuration M503 should give you the report

M503 has to do with EEPROM and even trying typing it in something like the WebUi and or Prontor face a 3d printer controller I get "unknown command"

mikedattolo, 2023-03-29:
Could it have to do with the Index.html.gz file?

mikedattolo, 2023-03-29:
Hello?

DivingDuck, 2023-03-29:
This sounds like you do have no connection to your printer. Sure you connected the correct pins for rx and tx to your printer? 
rx and tx need to be crossed esp_rx pin to printer_tx pin and exp_tx pin to printer_rx pin.

mikedattolo, 2023-03-29:
> This sounds like you do have no connection to your printer. Sure you connected the correct pins for rx and tx to your printer? rx and tx need to be crossed esp_rx pin to printer_tx pin and exp_tx pin to printer_rx pin.

Its gotta be connected cause I can send movement commands through the esp3d and the printer moves

mikedattolo, 2023-03-29:
> Then you should see the response for M115 as well in the terminal window. What baud rate had you set for printer and esp3d?
https://user-images.githubusercontent.com/113625156/228542126-c53bbb28-2b14-47a4-a3c2-417170d1d3ec.PNG
there is no response for M115
and both esp3d and printer have a bud rate of 250000

DivingDuck, 2023-03-29:
this is weird. can you reduce the baud rate to 115200 for both? Just in case ...

mikedattolo, 2023-03-29:
when I get the chance I will, I am currently in class so ill let you know when I can

luc-github, 2023-03-29:
You tested the ESP board as standalone and it seems to send and receive message : https://github.com/luc-github/ESP3D/issues/907#issuecomment-1488682596
So your ESP Board is working and ESP3D is also sending /receiving message

More over

I have an octopus with socket 1.1 with ESP32 socket on it and it is working so Marlin communicate as well as ESP3D 2.x

So your issue is in Marlin configuration or in your  wiring, not in ESP3D

if you see in PrinterFace  when typing M503 `unknown command` it means accessing EEPROM/config is not enabled 
I also suggest you to check the `M115` command in pronterface to see what is your Firmware version as you did not provided in template 

Additionnaly I see you are using ESP3D in AP mode which is not good idea - AP mode is only for configuration to setup and join network, in normal usage it will be too busy answering all network requests to be responsive for other tasks

closed 2023-04-02 by luc-github