luc-github/ESP3D #908: Setup mode does not connect to Router, reverts to Access Point without explanation.

3D-Darryl, 2023-03-28:
**Describe the bug**
Using ESP32 with attached camera. Cannot connect to routers with valid credentials. The serial does not show multiple attempts, and reverts to AP mode after 15 seconds even though the boot delay is 25. There is no message to explain this behavior.
**To Reproduce**
Configure radio for "Setup" mode.  Set the boot delay to 25 seconds.  Configure the network for your router. Connect the serial to a computer terminal. Restart the ESP32.
Note boot message, 15 second pause, then AP mode is started.
**Expected behavior**
Either: 1.) Multiple messages as it repeatedly tries to connect for 25 seconds, then if it fails, revert to AP mode, or 2.)Message about setup mode (whatever that is) and delay for boot-delay time before reverting to AP.
**ESP3D Firmware:**
FW ver: 3.0.0.a222
- ESP3D-WebUI Version:[e.g. 2.1b14] 
- Wifi mode: Setup
- Flash method: Arduino Ide  
- Arduino ESP8266 package version: 3.0.2
- Any change done in FW: (none)

**Target Firmware:**
- Name: Marlin  2.0
- MCU: ESP32
- Name: AI Thinker
- Flash size: 4M

**Additional context**
The problem is that the meaning of "Setup" mode is not clear.  "Client" Is what I wanted and has the expected behavior including fallback to AP.

luc-github, 2023-03-28:
it is because boot is set to quiet by default : 
you can enable using [ESP150]
```
Get/Set display/set boot delay in ms / Verbose boot
[ESP150]<delay=time in milliseconds><verbose=ON/OFF>pwd=<admin password>
```
`[ESP150]verbose=ON`

if go to fallback it means it cannot connect (Wrong AP, Wrong password, signal too weak, restriction on AP itself to accept new client)

Setup mode is a mode that catch all requests to reach ESP easily, it is also an AP mode but it is not recommended to use for printing because it is heavily handling all network request so it is for ESP configuration only.

3D-Darryl, 2023-03-28:
So I am suggesting that in Setup mode there is a message on the serial
saying something like "Setup mode is for configuration and not recommended
for normal operation".
This would have cleared up my confusion.
Darryl

luc-github, 2023-03-28:
you are welcome to update / add missing information in documentation : https://esp3d.io/esp3d/v3.x/documentation/index.html

closed 2023-03-29 by luc-github