luc-github/ESP3D #911: [BUG] Communication Issues

NeuDan316, 2023-04-01:
Hey,

just made an account, please tell me if I'm not following any community guidelines or something.

So I followed Knutwurst's Guide on WiFi Installation. 
https://github.com/knutwurst/Marlin-2-0-x-Anycubic-i3-MEGA-S/wiki/WiFi-Installation-(english)

Everything worked out, up until I tried to communicate with the printer.

I read through some Issues that sounded similar and came to:
https://github.com/luc-github/ESP3D/issues/907#issue-1642088555

And there I got stuck at:
https://github.com/luc-github/ESP3D/issues/907#issuecomment-1487801616

[image]

If I write Hello, some questionmark symbols show up.

Also it seems that I can't write anything in the Monitor Terminal.

[ESP420] from Webui:

{"chip_id":"13469350","cpu":"160","freemem":"33.33 KB","SDK":"2.2.1(cfd48f3)","flash_size":"1.00 MB","update_size":"306.89 KB","spiffs_size":"228.20 KB","baud_rate":"500000","sleep_mode":"None","channel":"11","phy_mode":"11g","web_port":"80","data_port":"8888","active_mode":"AP (AA:48:FA:CD:86:A6)","ap_ssid":"ESP3D","ssid_visible":"Yes","ssid_authentication":"WPA","ssid_max_connections":"4","ssid_dhcp":"Started","ip":"192.168.0.1","gw":"192.168.0.1","msk":"255.255.255.0","connected_clients":[{"bssid":"90:E8:68:15:9A:FB","ip":"192.168.0.100"}],"disabled_mode":"STA (A8:48:FA:CD:86:A6) is disabled","captive_portal":"Enabled","ssdp":"Enabled","netbios":"Disabled","mdns":"Enabled","web_update":"Enabled","pin recovery":"Disabled","autentication":"Disabled","target_fw":"Marlin","M117_output":"Enabled","Notifications":"Disabled","Serial_output":"Enabled","Websocket_output":"Enabled","TCP_output":"Enabled","fw":"2.1.1.b8 ESP8266/8586"}

Sorry if I forgot any information, let me know if you need any more details.
Thanks in advance!

BR
Daniel

luc-github, 2023-04-01:
looks like baud rate are not same on ESP and printer for serial

luc-github, 2023-04-01:
I see you use a serial monitor 115200 when ESP seems 500000 so of course they won't talk

NeuDan316, 2023-04-01:
Thanks for the quick answer!
I'm sorry if I ask "dumb" questions, but it's like learning to walk for me..

[image]

So did I change it correctly now?

And how / why does this affect my Webui -> Printer communication?
Does that mean the fw from knutwurst has the wrong baudrate and thats why it's not working?

(sorry for the wrong label, didn't know how to change it)

NeuDan316, 2023-04-01:
Oh and also: i after that i typed "Hi" in webui and it showed.
Still no idea how to type something in the serial monitor tho.

luc-github, 2023-04-02:
I suggest you to study GCODE commands , webui send GCODE cimmands  to ESP board which transfer them to your printer by serial, then esp board read answer from serial and transmit to webui.
so wrong serial communication=>no command send/answer receive

luc-github, 2023-04-02:
now you can make your esp communicate with your printer board the rest is  Marlin configuration to be sure all GCODE commands used are enabled.
you have some learning curve to parse by reading documentation and experimeenting.
I close issue then

closed 2023-04-02 by luc-github