luc-github/ESP3D #913: [BUG]<When uploading a file, the file creation time does not calculate the time zone>

liqijian101, 2023-04-02:
When uploading a file, the file creation time does not calculate the time zone.
For example, my time zone is - 8, and the upload time is 10:00 on April 2, 2023. After the file is uploaded, the display time is 18:00 on April 2, 2023.

luc-github, 2023-04-03:
Indeed the set time was messy - I did not really tested - I pushed to later and of course was buggy - sorry  
1 - Internet time was never  actually enabled only webui set the time was enabled
2 - WebUI initialization time string had a typo so time was not set properly
3 - 

So you need to update ESP3D and ESP3D-WEBUI to get full working solution 

I have tested with my local time zone (+8) and now everything seems ok using internet time or manual time

DivingDuck, 2023-04-03:
Maybe an additional point. It would be nice to see a actual date and time info so that we can recognize when the setting is wrong, e.g on the about page.

luc-github, 2023-04-03:
you mean `[ESP140]NOW` ?

DivingDuck, 2023-04-03:
Yes, like this should be fine

DivingDuck, 2023-04-03:
yea, this is what I had seen in the code yesterday.

[ESP140]
srv1=192.168.178.1, srv2=169.254.1.1, srv3=194.25.134.196, zone=2, dst=YES

I'm just build new binaries to see your last update. Still think, it would be nice to have it in the about section as I don't need to remember all the commands, it is still more intuitive as this page show most of the important settings with one click (and I like it very much)
:)

luc-github, 2023-04-03:
You do not need to remember just type `[ESP0] `or even `[ESP0]140`
Ho you mean display date/time of ESP in WebUI  ? well I though about it  but the issue is the query is done once when you open page so it will weird to still display the same time after a while, and always refresh the time is waste of bandwidth

DivingDuck, 2023-04-03:
Maybe with click on refresh for asking once the infomation

luc-github, 2023-04-03:
```
[ESP140]NOW
;2023-04-03 16:07:39
[ESP800]
;FW version:3.0.0.a224
;FW target:unknown
;FW ID:0
;Setup:Enabled
;SD connection:none
;Serial protocol:Raw
;Authentication:Disabled
;Web Communication:Synchronous
;Web Socket IP:192.168.2.218
;Web Socket Port:81
;Hostname:esp3d
;WiFi mode:STA
;Web update:Enabled
;Flash File System:LittleFS
;Host Path:/
;Time:Auto
```

DivingDuck, 2023-04-03:
Oh, I forgot NOW...

I need to set it to utc+1 for getting the correct time, but it should be utc+2 as we have a summer / winter time change.

https://time.is/en/Germany

luc-github, 2023-04-03:
did you enabled daylight ?

DivingDuck, 2023-04-03:
Yes i had enable it, but disable it shows the correct time. It should be yes.

I also struggle with this settings, to be honest. (I was just thinking to to make new backups for my applications with DS18B20 sensors because they show the correct time and I hadn't build the applications new now for quite some time. This was a friendly reminder for it ... )

Here is my setting for now displaying the correct time:
```
[ESP140]
srv1=192.168.178.1, srv2=169.254.1.1, srv3=194.25.134.196, zone=2, dst=NO
[ESP140] NOW
2023-04-03 11:02:07
```

luc-github, 2023-04-03:
>It should be yes.

If you set to Yes it add extra hour , so this setting fry my brain to be honest, because set to 0 does not apply the DST*3600
in [sample](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Time/SimpleTime/SimpleTime.ino#L54) esp32 and esp8266 have same behavior
and code is [here](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c#L20-L61)

I will wait winter to see what is the NTP server response and how DST affect the time

luc-github, 2023-04-04:
@liqijian101 does it fix the issue on your side ?

liqijian101, 2023-04-04:
I have just tested the new code and can now display the correct time. Thanks.

luc-github, 2023-04-04:
thank you for the feedback, I close issue then

closed 2023-04-04 by luc-github