luc-github/ESP3D #915: [BUG]<Build ESP32-C3 Project shows 'Serial2' was not declared in this scope>

tongsuh, 2023-04-15:
**Describe the bug**
i am new to ESP3D, and i downloaded the ESP3D-3.00-alpha3 RELEASE , using PlatformIO to build  ESP32-C3 Project Task.

TERMINAL shows :
esp3d/src/modules/serial/serial_service.cpp:41:59: error: 'Serial2' was not declared in this scope

ask for help, thanks a lot :)

PS: i didn't change any code. 

**To Reproduce**

1. Open 'VScode'->Open 'PlatformIO'->Open 'ESP3D-3.0.0-alpha3' folder->Open 'PROJECT TASKS'
4. Click on 'esp32-c3'->Click on 'General'->Click on 'build'
5. See error on TERMINAL 

**Expected behavior**

build SUCCESS

**Screenshots**

[image]

**ESP3D Firmware:**

- ESP3D FW version: 3.0.0-alpha3
- Flash method: PlatformIO
- Any change done in FW: changed NOTHING

luc-github, 2023-04-15:
please use latest git , not prerelease package,  this issue has already been fixed in it

tongsuh, 2023-04-15:
> please use latest git , not prerelease package, this issue has already been fixed in it : #825

thankyou , i'll have a try

tongsuh, 2023-04-16:
appreciate for your help, my problem fixed ,and also for those who download Releases , you can change the MAX_SERIAL value to 2 ,and delete '&Serial2' in serial_service.cpp
[image]

closed 2023-04-16 by tongsuh