luc-github/ESP3D #916: [FEATURE REQUEST] USB Host Mode

probonopd, 2023-04-16:
It would be nice if we could attach 3D printers to the ESP32S2 using USB Host Mode.

There are now very inexpensive ESP32-S2 based boards called the __S2 Mini__.

[image]

Photo credits: smartin015

They have a USB-C port which a 3D printer can be plugged in using a USB-A to USB-C adapter.

And Espressif seems to have a USB host mode driver for serial bridges and for CDC class compliant devices.

* https://components.espressif.com/components/espressif/usb_host_vcp Virtual COM Port Service, a wrapper around the following:
* https://components.espressif.com/components/espressif/usb_host_ftdi_vcp FTDI UART-USB converters driver
* https://components.espressif.com/components/espressif/usb_host_ch34x_vcp CH34x USB-UART converter driver
* https://components.espressif.com/components/espressif/usb_host_cp210x_vcp Silicon Labs CP210x USB-UART converter driver
* https://components.espressif.com/components/espressif/usb_host_cdc_acm USB Host CDC-ACM Class Driver

**Video Demo**:
https://www.youtube.com/watch?v=zkpkUem1ZGw

Maybe it can be integrated into ESP3D?

luc-github, 2023-04-16:
the automatic driver selection is only available since december in idf 5.1 which is not yet available in esp32-arduino yet, but should come soon.

I have already implemented the feature in https://github.com/luc-github/ESP3D-TFT which use idf 5.1 framework instead of arduino framework
 
so be part of esp3d based on arduino framefork is just matter of time

luc-github, 2023-04-16:
Also yhe FR is already addressed here: https://github.com/luc-github/ESP3D/issues/851

closed 2023-04-16 by luc-github