luc-github/ESP3D #917: Cannot print from SD card attached to ESP32-CAM

probonopd, 2023-04-17:
**Describe the bug**

Cannot print from SD card attached to ESP32-CAM.

I can upload a GCODE file to microSD (cumbersome).

But then, when I click the Play button, all I get is:

```
M23 /Form-Kubus_0.6n_0.2mm_PLA_MK2S_36m.gcode
M24
ok T:17.5 /0.0 T0:17.5 /0.0 @:0 B@:0
echo:Now fresh file: /Form-Kubus_0.6n_0.2mm_PLA_MK2S_36m.gcode
open failed, File: Form-Kubus_0.6n_0.2mm_PLA_MK2S_36m.gcode.ok
ok
```

Same with a short filename:

```
M23 /test.gcode
M24
echo:Now fresh file: /test.gcode
open failed, File: test.gcode.ok
ok
```

However, when I download the GCODE file, then I see that the file is OK.
Why does it say "open failed" when I try to print it?

It seems like it sends M23 which would require the file to be on the printer's SD card. But the file is on ESP3D's SD card. It needs to be streamed from there via the serial port.
I have `#define SD_DEVICE_CONNECTION ESP_DIRECT_SD` in `configuration.h`.

**To Reproduce**

1. Go to the web interface
2. Upload a GCODE to the SD card on the ESP32-CAM
3. Click "Play"
4. See error in the Terminal

**Expected behavior**

The CCODE is printed on the 3D printer

**ESP3D Firmware:**

FW version 3.0.0-a35.M2.1
WebUI version 3.0.0.a224

**Target Firmware:**

```
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:3D Printer EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
ok
```

**Board used (please complete the following information):**

ESP32-CAM

luc-github, 2023-04-17:
That normal the SD card of esp32cam ia not connected to Printer - there is no way the printer can access the files on it and print them:
https://github.com/luc-github/ESP3D/discussions/540
http://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html

the streaming is under dev on ESP32 but it is not ready yet

luc-github, 2023-04-17:
I close it as it is kind of duplicate of https://github.com/luc-github/ESP3D/issues/243

closed 2023-04-17 by luc-github