luc-github/ESP3D #919: [BUG] Not all services are announced with Zeroconf

probonopd, 2023-04-17:
**Describe the bug**

Zeroconf does not announce the printer as `_http._tcp`, only `_esp3d._tcp` (what software uses that, actually?) and `_arduino._tcp` (for OTA). As a result, I cannot open the web interface from the browser using Zeroconf. 

Also, `_ftp._tcp` is not advertised. As a result, I cannot access FTP via Zeroconf. 

Also, all other services are not advertised.

Why?

**To Reproduce**

1. Enable Zeroconf
2. On a computer, run a Zeroconf browser, e.g., `avahi-browse -a -r`. You should see `_http._tcp` and one entry per service (e.g., FTP,...) but they are missing

**Expected behavior**

HTTP and all services are advertised with Zeroconf so that Zeroconf-enabled applications can find them.

**ESP3D Firmware:**

3.0.0

**Board used (please complete the following information):**

ESP32-CAM

luc-github, 2023-04-17:
>_esp3d._tcp (what software uses that, actually?)

ESP3D it self :
Get available ESP3D list
output is JSON or plain text according parameter
[ESP450]json=<no> <pwd=admin/user>

luc-github, 2023-04-17:
about list of services, I did not implemented that yet - it is in todo list

luc-github, 2023-04-18:
http/ftp/telnet/webdav/websocket service declarations in mdns service added in a225

closed 2023-04-18 by luc-github

probonopd, 2023-04-18:
Thank you very much @luc-github. Appreciate it. :+1: