luc-github/ESP3D #921: [FEATURE REQUEST] ESP32 built-in temperature sensor support

ansonl, 2023-04-23:
**Is your feature request related to a problem? Please describe.**

No

**Describe the solution you'd like**

Ability to use ESP32 built in temperature sensor as one of the temperature sources. It would be helpful for getting an approximation of controller board temperature to dynamically turn on controller board cooling fans. 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/temp_sensor.html

**Describe alternatives you've considered**

I could wire and mount separate temperature sensor for the controller board and read from it in controller board firmware. It would be helpful to use the built in capability of ESP32 if it already exists. 

**Additional context**

luc-github, 2023-04-23:
this sensor is already used and temperature is displayed in information page 
[image]

ansonl, 2023-04-25:
Thank you for the info. That answered my question!

closed 2023-04-25 by ansonl