luc-github/ESP3D #923: ftp file transfer does not work - ESP3D V3.0

pagabee, 2023-05-06:
The ftp file transfer does not work, neither the upload nor the download.
I tried it with admin user and without authentication.

In the feature request #383 it was mentioned that file transfer via ftp is faster compared to the browser. I was curious, I wanted to try it.

Upload:
[image]

After reconnecting:
[image]

Download:
[image]

[ESP420]
chip id: 10338
CPU Freq: 240Mhz
CPU Temp: 62.8C
free mem: 175.96 KB
SDK: v4.4.3
flash size: 4.00 MB
size for update: 1.25 MB
FS type: LittleFS
FS usage: 104.00 KB/1.37 MB
baud: 115200
sleep mode: none
wifi: ON
hostname: esp3d
HTTP port: 80
Telnet port: 23
WebDav port: 8181
Ftp ports: 21,20,55600
sta: ON
mac: 94:E6:86:3B:62:28
SSID: DIGI-XKH8
signal: 98%
phy mode: 11n
channel: 11
ip mode: dhcp
ip: 192.168.100.50
gw: 192.168.100.1
msk: 255.255.255.0
DNS: 192.168.100.1
ap: OFF
mac: 94:E6:86:3B:62:29
i-time: ON
serial: ON
authentication: ON
M117: ON
notification: ON (none)
sd: shared (SDFat - 2.1.2)
SD updater: ON
Target Fw: marlin
FW ver: 3.0.0.a225
FW arch: ESP32

Hardwer: ESP32 D1 mini

luc-github, 2023-05-06:
Did you followed FTP client  configuration instruction to allow only one connection at once ?

http://esp3d.io/esp3d/v3.x/documentation/ftp/index.html

pagabee, 2023-05-06:
Yes, only one connection is allowed.
I formatted the SD card again to be safe.
The upload via browser is ok.

Now with FileZilla:
[image]
[image]

Upload:
[image]

Download:
[image]
Only the first kilobyte is copied.

luc-github, 2023-05-06:
Please share your ESP3D configuration.h 
Also why did you set : SD as shared ? with ESP32 D1 mini you can only use direct SD

pagabee, 2023-05-06:
The configuration file:
[configuration.zip](https://github.com/luc-github/ESP3D/files/11412449/configuration.zip)

I designed and made a shield for the ESP32 D1 mini. It contains a FST3257 chip, which switches the SPI lines, the other lines for the SDIO mode are hard wired. #823 

I am currently testing the circuit. It works with SPI mode (SDFat library) but it isn't work with SDIO 1 bit and 4 bit mode. _(This would be another topic..., maybe the 47k resistorvalues are to much, or the SD card does not support this mode, or switching between SPI and SDIO is not possible, I don't know yet...)_
[ESP3D_board.pdf](https://github.com/luc-github/ESP3D/files/11412432/ESP3D_board.pdf)

[image]
[image]

The SD card was formatted with a tool, not with Windows:
[image]

luc-github, 2023-05-06:
About formating you should use this tool : http://esp3d.io/tools/sdtools/index.html

I never tested WiSCP as FTP client only filezilla - I see you use default mode which use passive mode - I have tested using port 21
also I see you enabled WebDav can you check with WebDav disabled ?

pagabee, 2023-05-06:
I formatted the SD card with the tool you recommended.
The port is 21.
WebDav is disabled.
But the problem is the sama...

pagabee, 2023-05-06:
As is see it, the [SDIO interface has fixed](https://www.instructables.com/Select-SD-Interface-for-ESP32/) pins, so my circuit does not support it.

luc-github, 2023-05-06:
no it is customizable https://github.com/luc-github/ESP3D/blob/3.0/esp3d/src/modules/filesystem/sd/sdio_esp32.cpp#L105
give a try to https://luc-github.github.io

pagabee, 2023-05-06:
I changed the 47k pullup resistors to 10k, beacuse [10k is recommended](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html), but the ESP3D doesn't see the SD card in SDIO mode.
I have tested both 1GB and 32GB SD cards from different brand, too.

[image]

[configuration SDIO.zip](https://github.com/luc-github/ESP3D/files/11412970/configuration.SDIO.zip)

luc-github, 2023-05-06:
SDIO is working with the ESP32CAM I have
I suggest you to try with basic arduino code : https://github.com/espressif/arduino-esp32/tree/master/libraries/SD_MMC/examples/SDMMC_Test

luc-github, 2023-05-06:
I have just tested on esp32 board with SPI SD using filezilla and WinSCP and no issue so for

[image]
[image]

I used direct SD as I do not have your hardware but it should be transparent with shared SD 

So Issue is with shared SD or your hardware implementation of sharing SD 
In both case I cannot duplicate on my side

pagabee, 2023-05-06:
I'm tired today, I'll continue tomorrow...

pagabee, 2023-05-06:
Do you use Visual Studio Code or Arduino?

luc-github, 2023-05-06:
Visual Studio Code with PIO

luc-github, 2023-05-07:
I have just updated to release ESP32 core 2.0.8 (PIO 6.2.0) and still ok

pagabee, 2023-05-07:
Can you please build and send me the _firmware.bin_ file?
Config file: [configuration.zip](https://github.com/luc-github/ESP3D/files/11414091/configuration.zip)

Thanks in advance!

If that doesn't help either, I have to order another ESP module and SD card module...

luc-github, 2023-05-07:
I do not build binary sorry - if I start to do, everyone will ask for it so I do not do.

About your issue I do not see how change SD Card reader or ESP board would solve your issue, unless basic script is not working, but it would be limited to SD Card reader eventually if basic script is not working not esp board

I suggest you to test your hardware first will basic scripts to validate their capabilities and basic features

pagabee, 2023-05-10:
I did some tests. I hooked up a SD card slot directly to the ESP32 module.

SDFat is ok:
Browser upload speed: ~115 KB/s
Browser download speed: ~160KB/s
FTP upload speed: ~60KB/s
FTP download speed: ~63KB/s

It is interesting that the FTP transfer speed is slower.
At the beginning of the transfer the speed is about ~160KB/s and later it drops to ~50KB/s.

SDIO:
Does not work, I get always errorcodes in the logs. It doesn't matter if there is pull-up resistors or not. I will try it again later, I have to read about the problem on the internet...

I think that this issue can be closed, because it is not related to the SW.
Thank you for your support!

luc-github, 2023-05-11:
So issue is with your sharing board ?

About speed it may depend of files size also - I forget the performances with FTP TBH, I know webserver is not good on esp32 but webdav which does not rely on webserver multipart parsing  is better : https://github.com/luc-github/ESP3D/issues/576

I close issue then - good luck in your investigations

closed 2023-05-11 by luc-github

pagabee, 2023-05-13:
Hello Luc,

I was interested in why the ESP3D V3.0 does not see the SD card via SDIO, and did a little search on the internet. I want to share my experiences...
I found that, the ESP32 variant (released in 2016) has dedicated SDIO, so I used them: D2 - 12; D3 - 13; CMD - 15; CLK - 14; D0 - 2; D1 - 4.

The D2 and the D12 pins have special features, if they are pulled up for example by the pullup resistors of the SD card, the programming and booting of the ESP32 are unsuccessful. This caused me a lot headache...
That means, during the programming the D2 and D12 wires must be unconnected and during the booting the D12 must be unconnected. Otherwise the programming fails or the ESP32 does not boot.

https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/tutorials/io_mux.html
https://github.com/espressif/esp-idf/issues/2478
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino
 
I tested the SDIO upload speed in the browser, which is not as fast as I expected before compared to SPI:

ESP32:
SDIO4 upload speed: 116KB/s
SDIO4 download: 185KB/s
SPI upload speed: 115KB/s
SPI download speed: 160KB/s

Interestingly the upload speed is faster on ESP8266:
SPI upload speed: 152KB/s
SPI download speed: 102KB/s
I think that the ESP8266 libraries are better optimized.

In all cases the SD card, its formatting and the transferred file was the same (3,62MB). The SD_DETECT_PIN was used to detect the SD card. Without this the SDIO speed was ~6 times slower.
I noticed that if the sector size the SD card is larger, the copying speed is faster.

For me, the FTP does not work at all with SDIO.

About my sharing board...
It can be used with SPI mode and browser upload.
If the SPI mode and FTP upload speed or the SDIO transfer speed were higher, I would desing a new board.
Unfortunatelly the board is not compatible with the ESP8266 D1 mini board, because I tought that the upload speed with ESP32 is higher, therefore I optimised the board for it.