luc-github/ESP3D #927: [FEATURE REQUEST]<Please add Pause/Resume and STOP button>

j-snijder, 2023-05-20:
**Is your feature request related to a problem? Please describe.**

Missing Pause/Resume and STOP button on web interface>

**Please add Pause/Resume and STOP button**

When push pause, printing pause, on push Resume it resumes
When Push STOP printer goes to immediate stop of print and cooles down

**Describe alternatives you've considered**

Alternative could be to add a clear instruction on how to add custom buttons that actually communicate with the printer.

**Additional context**

luc-github, 2023-05-20:
This feature is implemented in V3 - did you tried it ?
on 2.1 you can add it to macro buttons according your printer

j-snijder, 2023-05-20:
Hi,

I have a Zonestar Z9V5 with WiFi and your ESP fw on it.
Macro function is there but command is not recognized by Marlin fw on printer, although M25 and M24 are Marlin codes.

I target the ESP as the choice of 3 target (ESP - URI or SD) But is that the right target ?

luc-github, 2023-05-20:
what FW version are you using?

j-snijder, 2023-05-20:
FW version: 2.1.0.c26 ESP8266/8586
WebUI version: 2.1b70

copied this from info screen of ESP

luc-github, 2023-05-20:
2.1 you need to create file with commands inside the file stored on ESP Filesystem or SD

j-snijder, 2023-05-20:
The web UI has an option for fw update, how to do it and where to find file ?

j-snijder, 2023-05-20:
<html>
<body>
<div class="modal-header" style="box-sizing: border-box; padding: 2px 16px; color: rgb(15, 15, 15); background-color: rgb(242, 242, 242); border-top-left-radius: 10px; border-top-right-radius: 10px; border-bottom: 1px solid rgb(207, 207, 207); font-family: Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><h3 style="box-sizing: border-box; font-family: inherit; font-weight: 500; line-height: 1.1; color: inherit; margin-top: 20px; margin-bottom: 10px; font-size: 24px;"><div class="modal-title" id="fw_update_dlg_title" translate="" english_content="ESP3D Firmware Update" style="box-sizing: border-box;">ESP3D Firmware Update</div></h3></div><div class="modal-body" style="box-sizing: border-box; padding: 10px 16px; color: rgb(51, 51, 51); font-family: Arial, sans-serif; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">

Select fileNo file chosen | Select file | No file chosen |   |   |   |   |  
-- | -- | -- | -- | -- | -- | -- | --
Select file | No file chosen

</div>
</body>
</html>

luc-github, 2023-05-20:
>The web UI has an option for fw update, how to do it and where to find file ?

you need to build the binary by yourself

j-snijder, 2023-05-20:
OKay, not for me then, am not into that kitchen skills

luc-github, 2023-05-20:
Have a look to https://esp3d.io/ it is not that hard if you use platformio - I have setup everything

j-snijder, 2023-05-20:
Can you then give me an example of the file I need to create on SD or upload to the ESP to get me a pause button and a resume button ?

All else what I need I can then add later

luc-github, 2023-05-20:
just a file with the gcode pause command / resume command  - if you are using Marlin 
Pause : https://marlinfw.org/docs/gcode/M025.html
Restart: https://marlinfw.org/docs/gcode/M024.html
Stop: https://marlinfw.org/docs/gcode/M524.html

luc-github, 2023-05-20:
As no new feature come to 2.X and this feature is already present in 3.0 I close FR

closed 2023-05-20 by luc-github