luc-github/ESP3D #928: ESP32

asdino2, 2023-05-23:
Compile firmware ESP3D 2.1.X (2.1.2 give error from start as env not properly named. ) 
Renamed esp3d.bin
Write on microsd card , flash on SKR3 EZ board . Red light flashing aand stop. 
No update the firmware on ESP32 wifi module  
Did not take the firmware update.
I bought it as new but I see has already a configuration because I see the Wifi SSD , I connect to wifi but no communication.

Thank you

luc-github, 2023-05-23:
>Compile firmware ESP3D 2.1.X (2.1.2 give error from start as env not properly named. ) 
What do you mean ? the env:esp32dev is same for both 

please folow this template : https://github.com/luc-github/ESP3D/blob/2.1.x/.github/ISSUE_TEMPLATE/bug-report---problem.md to report an issue

asdino2, 2023-05-23:
Hi, thank you for your time . I compile finally the firmware but I am not able to flash on ESP3D with a microSD . I just wanted to know if is ok to update the firmware by microsd as by instructions from ESP3D webpage

luc-github, 2023-05-23:
can you link me where it is explained that you can update the Firmware by micro sd ?

asdino2, 2023-05-23:
It's at 4.3.13 in BTT SKR3 EZ manual

https://github.com/bigtreetech/SKR-3/blob/master/BIGTREETECH%20SKR%203%20EZ%20user%20manual.pdf

thanks

luc-github, 2023-05-23:
That part is made by Marlin  / BTT code not ESP3D - so I do not know what could be wrong

luc-github, 2023-05-23:
my guess is the bootloader initially  made to work with esp8266 - so flashing esp32 may need bootloader update