luc-github/ESP3D #931: [BUG]<Error compiling Esp3d.ino on MacOs>

aayes89, 2023-06-19:
I got these errors when compiling the esp3d.ino using Arduino IDE 2.0 on MacOS 13.6 to a NodeMCU v1.0.0 with 4mb, mainly from the use of virtual identification:
I remove all existents libraries and copy the ones included on the zip file but always got the same errors even if i change the Arduino IDE version

In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/debug_esp3d.cpp:21:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h: In member function 'bool ESP8266WiFiSTAClass::beginWPSConfig()':
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h💯20: error: 'WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool' was not declared in this scope
100 | return WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool();
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/debug_esp3d.cpp:21:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h: At global scope:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h:83:10: error: conflicting return type specified for 'virtual size_t WiFiClient::availableForWrite()'
83 | size_t availableForWrite();
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Stream.h:27,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/HardwareSerial.h:32,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Arduino.h:303,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:23,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/debug_esp3d.cpp:21:
/Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.h:80:21: note: overridden function is 'virtual int Print::availableForWrite()'
80 | virtual int availableForWrite() { return 0; }
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/benchmark.cpp:21:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h: In member function 'bool ESP8266WiFiSTAClass::beginWPSConfig()':
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h💯20: error: 'WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool' was not declared in this scope
100 | return WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool();
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/benchmark.cpp:21:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h: At global scope:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h:83:10: error: conflicting return type specified for 'virtual size_t WiFiClient::availableForWrite()'
83 | size_t availableForWrite();
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Stream.h:27,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/HardwareSerial.h:32,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Arduino.h:303,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:23,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/benchmark.cpp:21:
/Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.h:80:21: note: overridden function is 'virtual int Print::availableForWrite()'
80 | virtual int availableForWrite() { return 0; }
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/espcmd/ESP101.cpp:20:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h: In member function 'bool ESP8266WiFiSTAClass::beginWPSConfig()':
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h💯20: error: 'WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool' was not declared in this scope
100 | return WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool();
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/espcmd/ESP101.cpp:20:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h: At global scope:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h:83:10: error: conflicting return type specified for 'virtual size_t WiFiClient::availableForWrite()'
83 | size_t availableForWrite();
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Stream.h:27,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/HardwareSerial.h:32,
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Arduino.h:303,
[... 124 lines cut ...]
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:34,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/espcmd/ESP102.cpp:20:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h: In member function 'bool ESP8266WiFiSTAClass::beginWPSConfig()':
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.h💯20: error: 'WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool' was not declared in this scope
100 | return WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool();
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFi.h:39,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/hal.h:28,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:43,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/espcmd/ESP102.cpp:20:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h: At global scope:
/Users/localadmin/Documents/Arduino/libraries/ESP8266WiFi/src/WiFiClient.h:83:10: error: conflicting return type specified for 'virtual size_t WiFiClient::availableForWrite()'
83 | size_t availableForWrite();
| ^~~~~~~~~~~~~~~~~
In file included from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Stream.h:27,
[... cut ...]
from /Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Arduino.h:303,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/include/esp3d_config.h:23,
from /Users/localadmin/Downloads/ESP3D-3.0/esp3d/src/core/espcmd/ESP102.cpp:20:
/Users/localadmin/Library/Arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/Print.h:80:21: note: overridden function is 'virtual int Print::availableForWrite()'
80 | virtual int availableForWrite() { return 0; }
| ^~~~~~~~~~~~~~~~~

luc-github, 2023-06-19:
you did not followed the template ...
What esp8266 core version are you using ?

aayes89, 2023-06-20:
I solve the problem using ESP8266 2.7.4 Core and modifying the pyserial library config file

closed 2023-06-20 by aayes89