luc-github/ESP3D #937: [BUG]<ESP32CAM doesn't work with [ESP201]>

Salvatore805, 2023-08-03:
**ESP32CAM doesn't work with [ESP201]**

i'm trying to make screenshot in cura with the Command [ESP201]P0 V1 but the terminal answer is "error: Missing pin".
how i can solve?

luc-github, 2023-08-03:
[ESP171] not  [ESP201]
http://esp3d.io/esp3d/v3.x/documentation/commands/index.html

closed 2023-08-03 by luc-github

Salvatore805, 2023-08-03:
I mean that fromCURA I want to send a GCODE command that allows me to take a picture and save it in the SD memory of the ESP32CAM

luc-github, 2023-08-03:
read my comment and read the link I gave it answer you question

Salvatore805, 2023-08-03:
yes i try but this is the result
[image]

luc-github, 2023-08-03:
are you trolling ? did you read the command I gave you ? why use ESP701 ?

Salvatore805, 2023-08-03:
sorry, my translator changed the command. now sending this [esp171] the terminal tells me that the snapshot has been taken but I can't see it saved in the SD