luc-github/ESP3D #938: [BUG] Problem with ESP-12S resetting with WDT and exceptions

andrejrcarvalho, 2023-08-07:
**Describe the bug**

Hi, I'm having some problems while trying to setup the ESP-12S with the ESP3D. 

I want to use the ESP-12S on my MKS Robin Nano v3.1 and put the ESP3D firmware on it. To flash it, I used PlatformIO and made the wiring needed on a breadboard to test it. 
Sometimes I am able to connect to it via Wi-Fi, but most of the time the ESP resets itself via the watchdog reset.
In one of my attempts, I was able to start uploading the web UI to it, but I got an exception, and the upload failed.
I have a other ESP8266 around (the NodeMCU ESP8266 board), and I tried to make the setup on it, and all works fine.
I don't understand what the problem is with my EPS-12S.

Both of these are based on the same microprocessor, so why does it not work on the ESP-12S?

Can someone help me.

Above are some outputs of the error that I receive.

```
Exception (0):
epc1=0x402197d5 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffff780 end: 3fffffc0 offset: 01a0
3ffff920:  00000100 3ffff980 00000005 4021deb0  
3ffff930:  fffffffc e93824d0 9208d1c5 732eb466  
3ffff940:  1b06a3e7 d5c2bd4d d86596ac b0d68e04  
3ffff950:  31bd21f8 d3206d05 99f922ce 4da8578d  
3ffff960:  b6529d8f 64c34980 556ac5b0 a8295856  
3ffff970:  cc8e8610 cd0901b2 dcdd96ed 4d681856  
3ffff980:  2756d845 c359fc2f c1512398 6d7e4ece  
3ffff990:  82a1b414 a2b681da 1bb96f13 099de95c  
3ffff9a0:  86b3c8f9 332582c7 4e3c130e d0be64b2  
3ffff9b0:  d0d392c4 714713a8 5f602f1c 130ae720  
3ffff9c0:  bf087372 4a4147d6 a08be933 70654b5f  
3ffff9d0:  d215eda3 4065bd71 0dabb63e e114899d  
3ffff9e0:  0d6d5db3 53fa69c8 28a1c275 f45a2c58  
3ffff9f0:  d540eb0d 563c434a 94d06af0 6438f42b  
3ffffa00:  4d6b108d 593243f5 370ea430 5111af61  
3ffffa10:  354b4dca 9a48158e d751adf3 4e625144  
3ffffa20:  04e114e6 9d4a23e9 00002035 00000024  
3ffffa30:  00002000 3fff10c8 00000100 00000030  
3ffffa40:  3fff1184 00148000 3fff10c8 40223011  
3ffffa50:  00000100 3fff0f6c 00000000 3fff12c0  
3ffffa60:  00000000 00080000 5946893c c3de0bc9  
3ffffa70:  00000000 00000030 00002200 00000100  
3ffffa80:  3fff10c8 00048000 00000024 4022093c  
3ffffa90:  3ffffb04 00000080 00000000 00002e23  
3ffffaa0:  00148104 00002e22 00000000 0000ffff  
3ffffab0:  00000100 3fff0f6c 00000020 00000001  
3ffffac0:  3ffffb04 00148104 3fff10c8 40223110  
3ffffad0:  00000000 400042db 3fff1184 00000100  
3ffffae0:  3fff10c8 00048000 00000024 4022093c  
3ffffaf0:  00000000 00000080 00000000 00002e23  
3ffffb00:  00000001 00002e22 00000000 0000ffff  
3ffffb10:  3fff0f6c 00000001 00000020 00000100  
3ffffb20:  0000000f 00000000 0000000f 4021dd48  
3ffffb30:  3ffffc00 000000f8 000000f8 00000024  
3ffffb40:  00149000 3ffffba0 3fff10c8 40220d5d  
3ffffb50:  00000000 00000000 3ffffba4 00000005  
3ffffb60:  3ffffc00 00149000 3fff10c8 40223110  
3ffffb70:  3fff12ac 00000007 3fff1184 3fff14e8  
3ffffb80:  3fff3ed0 00000050 00000050 3fff10c8  
3ffffb90:  00000001 00000100 3ffffc00 40220f6d  
3ffffba0:  0000000f 00040024 3fff10c8 00000051  
3ffffbb0:  3fff0018 3fff4408 000000fb 00000000  
3ffffbc0:  00000001 00100000 00002000 00000000  
3ffffbd0:  3fff0598 3fff0560 3fff32dc 3fff10c8  
3ffffbe0:  00000004 3fff0e6c 3fff0b5c 40221b90  
3ffffbf0:  00000001 3ffffc08 3ffffc0a 3ffee0c8  
3ffffc00:  02280018 043800fc 0000048f 40100438  
3ffffc10:  00000538 00021800 00000228 00000000  
3ffffc20:  00000004 00000006 00000000 000000fb  
3ffffc30:  00000800 3fff3ed0 00000222 00000100  
3ffffc40:  40103c52 401037ef 3ffee0c8 3fff3ed0  
3ffffc50:  00021800 00000800 00021800 4021f34d  
3ffffc60:  00000132 00000031 7fffffff 00000002  
3ffffc70:  401025b3 00080000 00040000 3fff3ed0  
3ffffc80:  00000800 00021800 3fff10c8 4021fa98  
3ffffc90:  3fff0b5c 401022b0 3fffc200 00000022  
3ffffca0:  3fff0b5c 40103532 00040000 00000022  
3ffffcb0:  4023826c 00000030 0000001c 00000800  
3ffffcc0:  3fff3e9c 00000001 3fff31dc 4021d345  
3ffffcd0:  00000000 0000002f 3ffef93c 402249dd  
3ffffce0:  ffffffff 3fffc6fc 0000005c 4020a5a5  
3ffffcf0:  00000001 000000a7 3ffffdf0 00000030  
3ffffd00:  40213f35 3fff1a6c 40224398 f765ed35  
3ffffd10:  00000000 00000001 00000001 00000006  
3ffffd20:  3ffffd70 3fff2204 00000000 4021b444  
3ffffd30:  00000091 00000041 3ffffd70 3fff1a88  
3ffffd40:  40212410 3fff21dc 00000001 40212400  
3ffffd50:  00000000 3fff21dc 00000000 3fff1a88  
3ffffd60:  00000034 00000001 3fff21dc 4022417a  
3ffffd70:  00000000 00000000 3fff21dc 401000c9  
3ffffd80:  00000034 402122e4 3fff21dc 402131d5  
3ffffd90:  00000000 00000000 ffff1a44 4021aee0  
3ffffda0:  00000034 3fff21dc 3fff1a44 40213885  
3ffffdb0:  6c69662f 00007365 00ff315c 4020e000  
3ffffdc0:  402131a4 00000091 00000034 4020e07a  
3ffffdd0:  00000000 00000000 3fff1a6c 3ffffdf0  
3ffffde0:  00000034 00000091 3fff1a44 40213f43  
3ffffdf0:  00000000 00000000 ff707954 00000000  
3ffffe00:  00000000 ffff3150 3fff2ff4 000e000f  
3ffffe10:  ff000000 3fff1cf4 0010001f ff006e69  
3ffffe20:  00000000 00000000 00ffffff 3fff324c  
3ffffe30:  0009002f ffff3150 3fff2f9c 001e004f  
3ffffe40:  ffffffff 3fff3150 3fff1a44 4021380d  
3ffffe50:  3fff1a6c 3ffffec0 00000000 0000007c  
3ffffe60:  3ffffdf0 00028b8b 402131a4 402122e4  
3ffffe70:  3ffe8a1d 3ffffea8 3fff1a44 00028b8b  
3ffffe80:  3ffe8a1d 3ffffea8 3fff1a44 4021489f  
3ffffe90:  00000000 00000000 00676e61 00000000  
3ffffea0:  00000000 fffef93c 3fff3b1c 000e006f  
3ffffeb0:  ff000006 3fff30e4 000f000f ff006e6f  
3ffffec0:  3fff30fc 0026002f ff000001 00000000  
3ffffed0:  00000000 00fef93c 00000000 00000000  
3ffffee0:  00ffffff 00000031 00000000 00fe8550  
3ffffef0:  6c69662f 00007365 00fe8550 54534f50  
3fffff00:  00000000 000000b4 3fff2f14 0000007f  
3fffff10:  ff73656c 4bc6a7f0 0001b2e7 3ffeff44  
3fffff20:  3fff1a6c 00000001 00000000 00000001  
3fffff30:  00000000 3fff1af8 00000000 40212410  
3fffff40:  00000000 3fff315c 3fff1a44 3ffeff44  
3fffff50:  00000001 3fff1a6c 3fff1a44 4021345f  
3fffff60:  40225108 00000000 00001388 402166a6  
3fffff70:  00000000 3fff315c 3ffef964 40214cbe  
3fffff80:  3ffef93c 00000000 3ffef964 40207007  
3fffff90:  3fffdad0 00000000 3ffeff14 40207090  
3fffffa0:  3fffdad0 00000000 3ffeff14 4021c138  
3fffffb0:  feefeffe feefeffe 3ffe8550 4010063d  
<<<stack<<<

 ets Jan  8 2013,rst cause:4, boot mode:(3,3)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

**ESP3D Firmware:**

- ESP3D FW version: 15f8078873458b0d6c6b224c69d406e45e808556
- ESP3D-WebUI Version:9161f4073401f23a31b9ac819490f37d62d7ef27
- Wifi mode:AP
- Flash method: PlatformIO

**Board used:**

- MCU:ESP-12S
- Flash size: 4M

luc-github, 2023-08-07:
for robin better use the ESP3D 3.0

There is no difference of code between target so it is may likely be a hardware issue - you mentionned it failed during upload so it is not FW related, but such issue can lead to upload a corrupted FW

What env did you used with 12S in platformio ?
What change did you do in config.h, platformio.ini or anywhere else ?

andrejrcarvalho, 2023-08-09:
The reset happens when I upload the index.tar.gz file to the ESP.

I did not change the any file I uses the stock configuration.

luc-github, 2023-08-09:
what the upload page look like ? do a screenshot please - I need to see the filesystem information

luc-github, 2023-08-10:
@andrejrcarvalho any update ?

andrejrcarvalho, 2023-08-10:
Hi again now I can not get it to boot up properly.
I think that this is a problem with the module itself.

When I reset the module by pulling the reset pin to low, the output of the module is:
```
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
rl��r��c�n��
            �
             ��p�|����x��ǒ��lp

                              �nn��;�n����

                                          b�cl`n���
                                                   ����l����
                                                            �
                                                             b�n��n�$�
                                                                      �
                                                                       b��>~�n�����l`��#�n�rnr���;��
                                                                                                    �
                                                                                                     �{r�ےn��b��`��x�l��x��r��n�b

�rl��r��c�n��
             �
              ��p�|����x��ǒ��p

                              �nn��;�n����
                                          �
                                           b�$rrp�n���
                                                      ����l����
                                                               �
                                                                b�n��n�$�
                                                                         �
                                                                          b��>~�n�����l`��#�n�rnr���;��
                                                                                                       �
                                                                                                        �{r�ےn��b��`��x�l��x��r��n�b�
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

```

And it gets stuck here forever. I am out of ideas to make this thing to work.
I'm thinking to get a new ESP module and replace this with a standard ESP8266.
I don't know....

luc-github, 2023-08-10:
I would suggest your to reflash but before erase the full flash it should solve the issues

Here the menu to use

[image]

luc-github, 2023-08-14:
@andrejrcarvalho  did the erase solved your issu?

luc-github, 2023-08-24:
close as hardware issue due to lack of response

closed 2023-08-24 by luc-github