luc-github/ESP3D #940: [BUG]<platformio.ini env:esp01s_160mhz] has a typo in the platform setting causing build to fail>

andy-spoo, 2023-08-11:
**Describe the bug**

**To Reproduce**

1. Select Visual Studio Code as the build environment.
2. Load V3 of ESP3D
3. Go to the Platformio section.
4. Select esp01s_160mhz as the build environment.
5. Build fails.

**Expected behaviour**

It should compile.

**EASY FIX**
Original code:-   platform = espressif8266@@4.1.0
Needs to have an '@' removed, so it reads platform = espressif8266@4.1.0

luc-github, 2023-08-11:
Indeed this typo is present since february and it is not seen by CI because it only test the esp8266 env not all 
Fixed in  https://github.com/luc-github/ESP3D/commit/7427f696b42f99352d9a07aa4ca4db2ba302dfdd

closed 2023-08-11 by luc-github

andyspoo, 2023-08-12:
It did make me feel like perhaps I am the only one using this board :-)

luc-github, 2023-08-12:
Well several people use esp32 / ESP12 or esp01 but with 4MB to have the webupdate