luc-github/ESP3D #949: [Question]<"What is the impact of the SPI flash size on the use of ESP32 firmware?>

liu2-3zhi, 2023-09-28:
"What is the impact of the SPI flash size on the use of ESP32 firmware? What are the differences between the ESP32-WROOM-32UE-N16 with a 16 MB SPI flash, the ESP32-WROOM-32UE-N8 with an 8 MB SPI flash, and the ESP32-WROOM-32UE-N4 with a 4 MB SPI flash? I noticed that the ESP32 flash is divided into two regions by ESP3D

luc-github, 2023-09-29:
ESP3D standalone Firmware  only need the 4MB Flash because the flash size partition that host the index.html.gz and related files can be set to a minimum of 250KB which let enough space for 2 APP for the updates even for large partitions needed by camera code for example
More flash is actually not really necessary on ESP32 for the moment

liu2-3zhi, 2023-09-29:
Thank you very much.

closed 2023-09-29 by liu2-3zhi