luc-github/ESP3D #950: WiFi support On Bugfix 2.1.x

JazzinSA, 2023-10-01:
I finally sorted out all warnings and errors when compiling by using Marlin Bugfix 2.1.x. Unfortunately there are a new issue now with the WiFi UI. 

The older version of marlin configured Wifi by simply enabling the WiFi in the adv config file, yet on the new bugfix V2.1.x the WebUI does not appear when I connect to the printer.

The old code looked like this:
/**
 * WiFi Support (Espressif ESP32 WiFi)
 */
//#define WIFISUPPORT         // Marlin embedded WiFi managenent
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

#if EITHER(WIFISUPPORT, ESP3D_WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery)
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host

Which means once I enabled ESP3D_WIFISUPPORT the websupport OTA support could also be enabled.
However the new Bugfix 2.1.x Looks different:

/**
 * Native ESP32 board with WiFi or add-on ESP32 WiFi-101 module
 */
//#define WIFISUPPORT         // Marlin embedded WiFi management. Not needed for simple WiFi serial port.
#define ESP3D_WIFISUPPORT   // ESP3D Library WiFi management (https://github.com/luc-github/ESP3DLib)

/**
 * Extras for an ESP32-based motherboard with WIFISUPPORT
 * These options don't apply to add-on WiFi modules based on ESP32 WiFi101.
 */
#if ENABLED(WIFISUPPORT)
  #define WEBSUPPORT          // Start a webserver (which may include auto-discovery) using SPIFFS
  #define OTASUPPORT          // Support over-the-air firmware updates
  #define WIFI_CUSTOM_COMMAND // Accept feature config commands (e.g., WiFi ESP3D) from the host

The result is that WEBSUPPORT and OTASUPPORT never gets enabled. And thus when I connect the web interface is missing. 

I can't find any other places to enable the ESP3D websupport. 
Any help will be appreciated greatly.

luc-github, 2023-10-01:
Enabling wifi on MARLIN is only for ESP32 board based where you install Marlin itself - nothing related to ESP3D
That is clearly stated in copy and paste you did 
WIFI_SUPPORT has nothing related with ESP3D neither
and ESP3D_WIFISUPPORT (NOT WIFISUPPORT) is for ESP3DLIB which is a library and it is not ESP3D FW which is is only for Daughter board not marlin, have a look to the diagrams here : http://esp3d.io/ for what is used for which purpose

JazzinSA, 2023-10-02:
Thanks luc for replying. I'm trying to compile the new version of Marlin for the Tiny-bee. I'll move the question over to the Tiny-bee

closed 2023-10-02 by JazzinSA

luc-github, 2023-10-02:
the setup of esp3dlib for marlin is explained here: https://github.com/luc-github/ESP3DLib#how-to-enable-
This will use the ESP3DLib 1.0

To use ESP3DLib V3 you need my fork because it use latest platformio  / arduino core which is not supported by Marlin yet
https://github.com/luc-github/ESP3DLib/tree/3.0