luc-github/ESP3D #951: [FEATURE REQUEST]<Please support Ethernet to spi modules similar to W5500>

liu2-3zhi, 2023-10-02:
**Is your feature request related to a problem? Please describe.**

The ESP32-S3 chip does not have a built-in MAC and PHY interface, so it cannot support Ethernet functionality directly like the normal ESP32.

**Describe the solution you'd like** 

Add support for external serial-to-ethernet chips such as W5500 and ENC28J60. This can be done by:

1. Implementing SPI driver for the ethernet chip. 

2. Implementing high-level ethernet driver based on the SPI driver.

3. Integrating the driver into ESP-IDF TCP/IP stack.

This will allow ESP32-S3 to use Ethernet functionality by connecting external ethernet chips.

**Describe alternatives you've considered**

Using third-party Ethernet libraries like Ethernet.h or LwIP. But this requires porting work and won't be able to leverage ESP-IDF native network stack.

**Additional context**

Example code for W5500 ethernet chip support on ESP32-S3:

```
// ethconfig.h

#include "driver/spi_master.h"
#include "w5500_ethernet.h"

class EthConfig {

  spi_device_handle_t spi_handle;

  bool init_w5500_spi() {
    // Init W5500 SPI 
  }

  bool linkUp() {
    // Check W5500 registers 
  }

};

// ethconfig.cpp

bool EthConfig::init_w5500_spi() {
  // Init SPI
  // Add device
  // Config SPI device 
} 

bool EthConfig::linkUp() {
  
  // Transmit SPI transaction
  // Check link status bit

}
```

This allows integrating W5500 to ESP-IDF native network stack.

luc-github, 2023-10-02:
The espressif idf seem already support ethernet on S3 for 1 year  (IDF 4.4.2) per this project https://github.com/W00ng/ESP32-S3-Gateway 
https://github.com/espressif/esp-idf/tree/master/examples/ethernet mention W5500 is supported

The idf code https://github.com/espressif/esp-idf/blob/master/examples/ethernet/enc28j60/README.md show enc28j60 being supported also

On arduino it may be already implemented - but because I do not have any S3 ethernet I did not tested neither see the issue of using current code

Once I will get one S3 device with W5500 - (I may order one https://github.com/W00ng/ESP32-S3-Gateway )I will have a look if there is a need to adapt ESP3D V3 code

luc-github, 2023-10-02:
Per this PR SPI ethernet is on going and will be in esp32 arduino core V3.0 : https://github.com/espressif/arduino-esp32/pull/8575 so should be  in coming months

liu2-3zhi, 2023-12-29:
> Per this PR SPI ethernet is on going and will be in esp32 arduino core V3.0 : [espressif/arduino-esp32#8575](https://github.com/espressif/arduino-esp32/pull/8575) so should be in coming months

It is finished now!

luc-github, 2023-12-29:
but 3.0 is not released yet

luc-github, 2024-09-03:
I am working on 5.1 porting and I have ordered 2 esp32-S3 with W5500 SPI modules (T-eth-lite)
I will try to order same modules with ESP32 , https://fr.aliexpress.com/item/1005006270867630.html

luc-github, 2024-09-05:
Support added but not yet tested due to lack of Hardware

closed 2024-09-05 by luc-github

luc-github, 2024-09-11:
Update made and tested with lilygo T-eth-lite ESP32-S3