luc-github/ESP3D #955: [FEATURE REQUEST]Change ESP_DIRECT_SD to ESP_ONLY_SD for consistency

luc-github, 2023-10-12:
in configuration.h
```
/* SD card connection
* ESP_NO_SD //(default)
* ESP_DIRECT_SD //Only your ESP board is connected to SDCard
* ESP_SHARED_SD //Printer SD Card is also connected to ESP3D
* Does your system has SD card and how it is connected to your ESP3D
*/
```
ESP_DIRECT_SD seems confusing for users and is often used when ESP_SHARED_SD should be used, the configurator must be updated also

TjeerdvanDijk, 2024-02-05:
Agreed, im trying to figure out what it means. 

currently it says `Your system has SD card and your ESP is connected to it using shared solution`

What exactly is a shared solution? Does it simply mean the printer has a sd card?

Sergey3112, 2024-05-16:
Как сделать, чтоб вообще видело SD карту?

closed 2024-11-25 by luc-github

luc-github, 2024-11-25:
Configurator and Documentation have been updated also

commit d2c52ac7 (2024-11-25): Close #955

Add more default settings for SPI W5500 Ethernet
Bump version for tracking