luc-github/ESP3D #957: [BUG] FTP not showing full directory listings on SD storage on Fysetc SD-WIFI-PRO

y2kdread, 2023-10-23:
**Describe the bug**

When I access esp3d through FTP, I am unable to list the full list of files/directories on the SD portion of the storage.

**To Reproduce**

1. Connect to FTP using FTP client
2. Browse to the "SD" folder
3. Full list of files/folders are not shown

**Expected behavior**

I should be able to see all files and directories on the SD card

**Screenshots**

[https://imgur.com/YGLZEmg](url)
[https://imgur.com/0UFANrS](url)

**ESP3D Firmware:**

- ESP3D FW version: 3.0.0.a225
- ESP3D-WebUI Version: 3.0.0-a40.M2.1
- Wifi mode: Client Mode
- Flash method: PlatformIO

**Board used (please complete the following information):**

- MCU: ESP32
- Name: Fysetc SD-WIFI-Pro
- Flash size: 8M

luc-github, 2023-10-24:
Did you configured your ftp client properly for esp3d ? 

https://esp3d.io/esp3d/v3.x/documentation/ftp/index.html

[image]

luc-github, 2023-10-24:
I cannot duplicate your issue using the setting as above
[image]

y2kdread, 2023-10-25:
That setting was incorrect in FileZilla, but I am still seeing the same thing with it set to only 1 connection.

[image]

luc-github, 2023-10-25:
this issue is different than webdav one ? even on limited number of file you have issue ?

luc-github, 2023-10-25:
looks like more the fat file system is corrupted in that case 
like not clean eject from pc

y2kdread, 2023-10-25:
I've actually tried testing this out by performing a full reflash of ESP and formatting the card with the SD formatting tool. I can try again just to be safe.

luc-github, 2023-10-25:
after formating the SD how did you uploaded files  ?

y2kdread, 2023-10-25:
I just copied the data from the original SD card.

luc-github, 2023-10-26:
my question was : HOW ?
Windows File manager ?
Your OS file manager ?
WebDav ?
your  CPAP machine ?

y2kdread, 2023-10-26:
Oh ok. 

I copied using Windows file explorer from the original card to the esp card.

luc-github, 2023-10-26:
I did same and I reproduce your issue 
[image]
which is weird is that webdav is able to see the content 
It has been years that I did not looked at this part of code I will dig in

luc-github, 2023-10-26:
Ok I am solving the bug - looks like the SD access was not well handled and sometimes even handled upside down...
So only first file/dir was listed on SD, no issue on FS because the FS is never released.

Listing 1000 files take around 4min in debug mode but so far it is working on my devt branch - I will continue to to review ftp code today and push the fix if no blocking point

luc-github, 2023-10-27:
Fix is now available - please give a try

y2kdread, 2023-10-27:
The fix you put in place resolved the issue. Thanks a ton!

closed 2023-10-27 by y2kdread