luc-github/ESP3D #959: [Question] Is the Creality V1.1.4 mainboard supported ?

tritri0903, 2023-11-05:
# Issue: Communication Problem Between ESP32 and Ender 3 (V1.1.4) (Bit Size Compatibility)

## Description

I am encountering difficulties in establishing communication between an ESP32 module and my Ender 3 printer with a V1.1.4 board. I suspect that the issue might be related to the differences in working bit sizes – the Ender 3 operates on 8 bits, while the ESP32 is a 32-bit device. This GitHub issue is to seek help, insights, and solutions to address this bit size compatibility problem.

## Current Situation

- Ender 3 3D printer with Marlin 1.1.9 firmware and a V1.1.4 board.
- ESP32CAM with ESP3D V3.0.
- Connected the two using UART with Rx and Tx, following the documentation.
- Attempts to establish communication between the two devices have been unsuccessful.

## Steps Taken

To address this issue, I have already taken the following steps:

- Checked the connection with a logic analyzer.
- Utilized both ESP3D V2.1 and V3.0.
- Tried both an ESP32 classic and an ESP32CAM module.
- Conducted research on relevant documentation and online resources, but found an issue (#363) without a solution.

## Environment

- Ender 3 3D printer with a V1.1.4 board (8-bit architecture).
- ESP32CAM module (32-bit architecture).

## Versions

- Ender 3 Marlin 1.1.9.
- ESP32CAM ESP3D V3.0 (and V2.1, with the same issue).

luc-github, 2023-11-05:
no that is not the issue a 8bits board can talk to 32bits by serial without issue - esp32 is sensitive and ony work on gpio pins with 3.3v - are your rx /tx using 3.3v or 5v  from your board ?

tritri0903, 2023-11-05:
Yes, I am aware of the 3.3V and 5V voltage issue with the RX and TX pins. Initially, I used a voltage divider to lower the voltage for the ESP32, but unfortunately, this didn't establish communication. To ensure voltage compatibility, I subsequently utilized a logic level converter, but the communication issue between the two devices persisted.

luc-github, 2023-11-05:
you connect to which uart on your board ? the one used by USB also ?

tritri0903, 2023-11-05:
I connected to the same pins as instructed in the tutorial (https://esp3d.io/esp3d/v3.x/installation/wiring/index.html#creality-cr10-ender-3-board). The connection to the UART of the board is confirmed to be correct because, using my logic analyzer, I can observe the Marlin startup message.

luc-github, 2023-11-05:
are you using same baud rate as Marlin ?

tritri0903, 2023-11-05:
Yes both are setup on 115200.

luc-github, 2023-11-05:
and nothing is connected to USB ?

tritri0903, 2023-11-05:
Nothing in the usb and in the micro-sd slot.

luc-github, 2023-11-05:
I would the suggest to find a USB/Serial adapter  like these : https://www.aliexpress.com/w/wholesale-ftdi-adapter.html
and connect its tx/rx/gnd to your rx/tx / gnd  of your board and connect it to a serial monitor and check you can communicate with your board to be sure rx / tx are ok for communication send M105 or M115 and get response 

If ok then do same with your esp32 and send [ESP420] or [ESP800] this will also allow you to be sure the pins are functionnal

That would help to debug your issue

tritri0903, 2023-11-05:
Thank you! I'll give this a try and will report back with the results.