luc-github/ESP3D #962: file upload always rejected

juergmaier, 2023-11-08:
SD-files on card not shown, upload new files gets rejected after showing progress bar.

I have formatted the sd card with fat32.
I am using an ESP32-CAM board.
I have run a script that reads, writes, appends data to the sd card and it reported file actions as successful.
I have created a test.txt file on the SD-card.

Using the ESP3D web view the SD card is shown as empty (the file test.txt is not shown)
Trying to upload a gcode file always gets rejected with error 12, sometimes immediately, somtimes after showing a partial progress bar.

ESP3D Firmware: 2.1.3b0

- Wifi mode: AP
- Flash method: [Arduino Ide 2.2.1]  

- Any change done in FW: None

Target Firmware: Marlin

Board Name: ESP32-CAM

Error on console:

11:29:15.819 -> M117 SD upload failed
11:29:17.819 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
11:29:17.819 -> Core 1 register dump:
11:29:17.819 -> PC      : 0x400e1744  PS      : 0x00060430  A0      : 0x800e1820  A1      : 0x3ffb1ad0  
11:29:17.819 -> A2      : 0x00000000  A3      : 0x3ffb1b1f  A4      : 0x00000001  A5      : 0x00000001  
11:29:17.819 -> A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
11:29:17.852 -> A10     : 0x3ffd21d8  A11     : 0x00000000  A12     : 0x3ffc1b4c  A13     : 0x3ffc74b8  
11:29:17.852 -> A14     : 0x3ffc1b40  A15     : 0x3ffc74b8  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
11:29:17.852 -> EXCVADDR: 0x00000008  LBEG    : 0x4008c149  LEND    : 0x4008c159  LCOUNT  : 0xffffffff  
11:29:17.852 -> 
11:29:17.852 -> ELF file SHA256: 0000000000000000
11:29:17.883 -> 
11:29:17.883 -> Backtrace: 0x400e1744:0x3ffb1ad0 0x400e181d:0x3ffb1af0 0x4018cc9d:0x3ffb1b10 0x400e3c41:0x3ffb1b40 0x400e4861:0x3ffb1b60 0x400e5759:0x3ffb1d60 0x400e6e79:0x3ffb1f20 0x400d82d9:0x3ffb1f70 0x400d833a:0x3ffb1f90 0x400ede08:0x3ffb1fb0 0x4008dde6:0x3ffb1fd0
11:29:17.883 -> 
11:29:17.883 -> Rebooting...

juergmaier, 2023-11-08:
> ESP32 Cam is not supported in ESP3D 2.1 also ESP3D 2.1 does not allow
> direct Sd access, so the card you try to acces is printer one using M28/M29
> protocol
> Is that the case ?

Hi Luc
Thanks for coming back with this.
Can you point me to a supported ESP board?
What version of esp3d do I need to access the Sd card?
I have no printer connected yet but plan to use it with anycubic kobra max.
It has a USB port and I can print over serial using Simplify 3d.
Regards
Juerg

juergmaier, 2023-11-08:
I have found this video about ESP32CAM and esp3d by Jason Winfield.
https://www.youtube.com/watch?v=CdHXRQlK3ds
Unfortunately my English is not good enough to be able to follow his vocal
explanations.

luc-github, 2023-11-09:
On ESP32 Cam the Connected SD is used to save frame from Camera, currently there is no way to `share` this SD with printer, so even you upload file on it the printer won't see it. Also ESP3D does not have streaming feature for 3D printer only basic one for CNC GRBL and even this one is not optimized for laser usage.
The https://github.com/luc-github/ESP3D/issues/243 give current status on ESP3D - and the SD usage is still valid here : https://github.com/luc-github/ESP3D/discussions/540#discussioncomment-2897564

juergmaier, 2023-11-09:
Hi Luc
Looks like you have a large task at hand.
I have downloaded esp3d version 3.0 (alpha) but get a compile error
(Arduino IDE 2.2.1) :

/home/juerg/Arduino/ESP3d/ESP3D-3.0/esp3d/src/modules/camera/camera.cpp: In
member function 'bool Camera::handle_snap(WebServer*, const char*, const
char*)':
/home/juerg/Arduino/ESP3d/ESP3D-3.0/esp3d/src/modules/camera/camera.cpp:81:5:
error: 'HTTP_Server' has not been declared
     HTTP_Server::set_http_headers();
     ^~~~~~~~~~~

exit status 1

Compilation error: 'HTTP_Server' has not been declared

Any hint?

Thanks
Juerg

luc-github, 2023-11-09:
it have been fixed this morning https://github.com/luc-github/ESP3D/issues/963 - you need to sync your download with latest code

juergmaier, 2023-11-10:
Downloaded your latest code just now. Compilation results now in:
/home/juerg/Arduino/ESP3d/ESP3D-3.0/ESP3D-3.0/esp3d/src/modules/http/handlers/handle-description_xml.cpp:
In static member function 'static void HTTP_Server::handle_SSDP()':
/home/juerg/Arduino/ESP3d/ESP3D-3.0/ESP3D-3.0/esp3d/src/modules/http/handlers/handle-description_xml.cpp:39:48:
error: 'class SSDPClass' has no member named 'getSchema'; did you mean
'schema'?
         _webserver->send(200, "text/xml", SSDP.getSchema());
                                                ^~~~~~~~~
                                                schema

exit status 1

Compilation error: 'class SSDPClass' has no member named 'getSchema'; did
you mean 'schema'?

luc-github, 2023-11-10:
are you using the correct version of SSDP  ? the one I provide in repository

juergmaier, 2023-11-10:
I have a folder ESP3D-3.0/libraries/ESP32SSDP-1.2.0 in my project.
In "ESP3D-3.0/esp3d/src/modules/network/netservices.cpp" an "#include
<ESP32SSDP.h"
Does it need to be in another location?

luc-github, 2023-11-10:
are you compiling using arduino or platformio ?
you error seems showing you compile from Arduino ide

juergmaier, 2023-11-10:
right, i am using Arduino IDE 2.2.1

luc-github, 2023-11-10:
then as mentioned in doc https://esp3d.io/esp3d/v3.x/installation/arduino/index.html
[image]

juergmaier, 2023-11-10:
Hi Luc
I followed your instructions and it's compiling now :-). Will soon try if
it works for me.
I hope I didn't distract you too much.

luc-github, 2023-11-10:
let me guess you:
 enabled pin reset on esp32cam?
or you enabled sd as spi card reader

juergmaier, 2023-11-10:
what would show more information during the setup?

my configuration.h:
/************************************
*
* SD card pins
*
************************************/

/* SD card detect pin
* The pin used to detect SD card
*/
// #define ESP_SD_DETECT_PIN 4

/* SD card detect pin value
* State of SD card detect pin when card is present
*/
// #define ESP_SD_DETECT_VALUE 0

/* SD shared flag pin
* The pin used to enable SD card for ESP board
*/
// #define ESP_FLAG_SHARED_SD_PIN -1

/* SD shared flag pin value
* State of SD card shared pin for ESP board
*/
// #define ESP_FLAG_SHARED_SD_VALUE 0

/* SD card CS pin
* The pin used to select SD card in SPI mode
*/
// #define ESP_SD_CS_PIN 5

/* Enable pin reset feature
* Use a pin to reset ESP3D settings
*/
// #define PIN_RESET_FEATURE

luc-github, 2023-11-10:
I doubt this is the  configuration.h 
most of configuration are missing

juergmaier, 2023-11-10:
it was just the part of configuration.h  I thought would be related to your
assumptions.

Does it matter what partition schema I use? I currently have selected "Huge
App (3 MB no OTA, 1 MB SPIFFS)"

I have tried to modify configuration.h to show more information:
#define ESP_LOG_FEATURE LOG_OUTPUT_SERIAL0

#define ESP3D_DEBUG_LEVEL LOG_LEVEL_VERBOSE

this is what I get in the serial monitor (repeated after 10s)
475:4

16:12:30.261 -> [ESP3D][boot_delay.cpp:54] begin(): Boot delay done

16:12:30.261 -> [ESP3D][settings_esp3d.cpp:337] read_byte(): read_byte 855

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1088]
getDefaultByteSetting(): getDefaultByteSetting 855

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1518] getSettingPtr(): Got index 855:

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1519] getSettingPtr(): type 0:

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1520] getSettingPtr(): size 1:

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1521] getSettingPtr(): default_val 1:

16:12:30.302 -> [ESP3D][settings_esp3d.cpp:1092]
getDefaultByteSetting(): getDefaultByteSetting found

16:12:30.325 -> [ESP3D][settings_esp3d.cpp:1094]
getDefaultByteSetting(): getDefaultByteSetting is 1

16:12:30.325 -> [ESP3D][settings_esp3d.cpp:339] read_byte(): default value 1

16:12:30.325 -> [ESP3D][esp_sd.cpp:160] accessFS(): Accessing Direct SD

16:12:30.325 -> [ESP3D][esp_sd.cpp:164] accessFS(): Checking SD state

16:12:30.325 -> [ESP3D][sdio_esp32.cpp:62] getState(): last init was
failed try sd_mmc begin

16:12:30.389 -> E (10855) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).

16:12:30.389 -> [ESP3D-ERROR][sdio_esp32.cpp:74] getState(): sd_mmc begin failed

16:12:30.389 -> [ESP3D-ERROR][esp_sd.cpp:166] accessFS(): SD not present

16:12:30.389 -> [ESP3D][esp_sd.cpp:181] releaseFS(): Release SD

16:12:30.389 -> [ESP3D][esp3d.cpp:96] begin(): Mode 0

16:12:30.421 -> [ESP3D][settings_esp3d.cpp:1518] getSettingPtr(): Got index 168:

16:12:30.421 -> [ESP3D][settings_esp3d.cpp:1519] getSettingPtr(): type 2:

16:12:30.421 -> [ESP3D][settings_esp3d.cpp:1520] getSettingPtr(): size 7:

16:12:30.421 -> [ESP3D][settings_esp3d.cpp:1521] getSettingPtr():
default_val ESP3D:

16:12:30.421 -> [ESP3D-ERROR][settings_esp3d.cpp:695]
GetSettingsVersion(): Invalid Settings Version

16:12:30.454 -> [ESP3D][esp3d.cpp:98] begin(): Need reset settings

16:12:30.454 -> [ESP3D][serial_service.cpp:408] reset(): Reset serial

[... 466 lines cut ...]
16:12:31.446 -> [ESP3D][settings_esp3d.cpp:1520] getSettingPtr(): size 4:

16:12:31.477 -> [ESP3D][settings_esp3d.cpp:1521] getSettingPtr():
default_val 115200:

16:12:31.477 -> [ESP3D][serial_service.cpp:159] begin(): Baud rate is
115200 , default is 115200

16:12:31.477 -> [ESP3D][serial_service.cpp:199] begin(): Serial 0 for
1 is started

16:12:31.477 -> ets Jul 29 2019 12:21:46

16:12:31.477 ->

16:12:31.477 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

16:12:31.477 -> configsip: 0, SPIWP:0xee

16:12:31.509 ->
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

16:12:31.509 -> mode:DIO, clock div:1

I have attached my complete configuration.h

juergmaier, 2023-11-10:
I have formatted the SD card with Fat32. now it gets detected but the app
is looking for an "ini" File?

16:53:02.032 -> [ESP3D][settings_esp3d.cpp:1092]
getDefaultByteSetting(): getDefaultByteSetting found

16:53:02.064 -> [ESP3D][settings_esp3d.cpp:1094]
getDefaultByteSetting(): getDefaultByteSetting is 1

16:53:02.064 -> [ESP3D][settings_esp3d.cpp:339] read_byte(): default value 1

16:53:02.064 -> [ESP3D][esp_sd.cpp:160] accessFS(): Accessing Direct SD

16:53:02.064 -> [ESP3D][esp_sd.cpp:164] accessFS(): Checking SD state

16:53:02.064 -> [ESP3D][sdio_esp32.cpp:62] getState(): last init was
failed try sd_mmc begin

16:53:02.226 -> [ESP3D][sdio_esp32.cpp:65] getState(): sd_mmc begin succeed

16:53:02.226 -> [ESP3D][sdio_esp32.cpp:69] getState(): sd_mmc card type succeed

16:53:02.226 -> [ESP3D][esp_sd.cpp:171] accessFS(): SD present

16:53:02.258 -> [ESP3D][esp_sd.cpp:173] accessFS(): Accessing SD is ok

16:53:02.258 -> [ESP3D][update_service.cpp:481] begin(): Update SD for
update requestest

16:53:02.258 -> [ESP3D-ERROR][esp_config_file.cpp:45] processFile(): No ini file

16:53:02.258 -> [ESP3D-ERROR][update_service.cpp:494] begin():
Processing ini file failed

16:53:02.258 -> [ESP3D][esp_sd.cpp:181] releaseFS(): Release SD

luc-github, 2023-11-11:
yes if you enabled SD updater
https://esp3d.io/esp3d/v3.x/documentation/update/index.html

luc-github, 2023-11-13:
@juergmaier I found a glitch in code that was introduced with recent PR that mess up the reset and in some occasion and  lead to dead loop reset at start
You may want to  try latest git to confirm that was your issue

juergmaier, 2023-11-13:
Hi Luc
Thanks for your notification.
I have loaded it and it looks much better, no resets any more
I do not understand the esp_core_dump error or how to correct this setting.
I also have the problem that IP address 192.168.0.1 is my routers ip.
I have tried to assign the ESP32 a different IP with the esp3dcnf.ini on
the vfat formatted SD card but it looks like the card is not accessed any
more (before it renamed the .ini file to .ok, also no SD related logs
appear on the serial monitor).
Maybe you have some more advice how to make it run?

6:45:25.123 -> ets Jul 29 2019 12:21:46

16:45:25.123 ->

16:45:25.123 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

16:45:25.123 -> configsip: 0, SPIWP:0xee

16:45:25.123 ->
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

16:45:25.123 -> mode:DIO, clock div:1

16:45:25.123 -> load:0x3fff0030,len:1344

16:45:25.123 -> load:0x40078000,len:13964

16:45:25.123 -> load:0x40080400,len:3600

16:45:25.123 -> entry 0x400805f0

16:45:25.542 -> E (418) esp_core_dump_f�� ͡: Incorrect size of core
dump image: 196611

16:45:36.094 -> M117 Setup wifi

16:45:36.803 -> M117 AP SSID: 'ESP3D' is started and protected by
password (setup mode)

16:45:38.811 -> M117 192.168.0.1

16:45:39.911 -> M117 192.168.0.1

luc-github, 2023-11-13:
the partition for ESP3Cam with ESP3D should be min_spiffs.csv https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/min_spiffs.csv,  I never had such message
About SD I do not know how you configured it in configuration.h so I have no clue
You can configure ESP3D using web interface or serial terminal by ESPdirect commands

I encourage you to read the doc : https://esp3d.io/esp3d/v3.x/documentation/commands/index.html

juergmaier, 2023-11-13:
sorry Luc, it looks to be over my head.
I was able to change the IP with [ESP107]192.168.0.200, shows with [ESP111]
After reboot this ip is shown in the log but I can't ping the address.
Maybe there are just too many options how to set up the ESP32.
To get the ESP3d shown in the browser I expected to set it into AP-Mode and
then access it in my browser with http://192.168.0.200
The SD card thing is also a problem to me. What should I do with the
content of the link you sent me (min_spiffs.csv)? Is there a tutorial how
to set up the SD-Card?

juergmaier, 2023-11-13:
Some progress now.
I was able to connect my WiFi to the SSID ESP3D.
I was also able to upload a file to the SD-Card (it still is vfat)
All the links however (Firmware/Interface/Help) result in 'unable to
connect'

luc-github, 2023-11-14:
are you in Access Point or Client Mode ?

juergmaier, 2023-11-14:
how would I know?

luc-github, 2023-11-14:
you do not know your own setup ?.....
So why willing to change the ip address ?
Access point mode : you connect to esp board wifi  with your PC to esp network

Client mode : the esp joint your local wifi and can access it thrue your local network

juergmaier, 2023-11-14:
Sorry, not very fluent in networking. Based on your clarification:
I am in Access Point mode and connect the Wifi on my PC to the ESP3D
network.

When I point my browser window to this address I get this page shown

[image: image.png]

I assume the File I see is on Flash and not on the SD card.
And as the links of Firmware, Interface and Help point to internet
addresses they do not work with my connection to the ESP3D?

09:09:40.834 -> E (439) esp_core_dump_f�� ͡: Incorrect size of core
dump image: 196611

09:09:51.404 -> M117 Setup wifi

09:09:52.145 -> M117 AP SSID: 'ESP3D' is started and protected by
password (setup mode)

09:09:54.175 -> M117 192.168.0.200

09:09:55.237 -> M117 192.168.0.200

09:09:59.106 -> M117 New client

luc-github, 2023-11-14:
well in access mode you are on limited network of esp so there no magic solution for you esp to access internet
The access point is made for configuration purpose not daily usage

you must make.the.esp to join your local network after you can access to internet

juergmaier, 2023-11-14:
another question:
I have tried to set the ESP into STA mode but can't find a command to
switch from AP to STA mode?
I have set the SSID with [ESP100] and mode to STATIC with [ESP102]
[ESP110] shows WIFI-SETUP

juergmaier, 2023-11-14:
> why change ip to static ? as you seems.not handling network per your
> feedback it is sireit won'tbe properly configured.
> [ESP110]WIFI-STA is the command to switch to Station mode

applied [ESP110]WIFI-STA
and [ESP102]DHCP
In the startuplog I assume Connecting fails and it falls back to AP mode?

1:59:59.272 -> M117 Setup wifi

11:59:59.400 -> M117 Connecting

12:00:20.081 -> M117 AP SSID: '192.168.0.201' is started and protected
by password (setup mode)

12:00:22.054 -> M117 192.168.0.201

12:00:23.156 -> M117 192.168.0.201

12:03:06.559 -> ;WIFI-SETUP

luc-github, 2023-11-14:
[ESP100] with your acces point  SSID
[ESP101] with your acces point password
if any of them is wrong so ESP3D cannot join wifi it fall back to ESP3D access point mode to let you fix your issue

juergmaier, 2023-11-14:
new situation
I switched to use jason's configuration
https://jasonwinfield.nz/esp3d-config-file/
I compiled and loaded again.
I used ESP100 and ESP101 to set my SSID and pwd
esp3d started and it found a free IP.
I can ping that IP and my router shows it as connected device.
Entering this IP in my browser however tells me that it is unable to
connect?

luc-github, 2023-11-14:
what is your opereating system ?

juergmaier, 2023-11-14:
ubuntu 23.10

luc-github, 2023-11-14:
esp3d is discoverable if you do not  know the ip using one of these tools :
https://esp3d.io/tools/discovery/

juergmaier, 2023-11-14:
could not make these tools to install/run
I have tried the browser Falkon and it says "192.168.0.115 refused to
connect"
Do I need another setting in the ESP3d to make the site accessible?

luc-github, 2023-11-15:
check what [ESP420] give as output

juergmaier, 2023-11-15:
08:42:25.527 -> ;echo:chip id: 20694

08:42:25.527 -> ;echo:CPU Freq: 240Mhz

08:42:25.527 -> ;echo:CPU Temp: 61.1C

08:42:25.527 -> ;echo:free mem: 106.95 KB - PSRAM:3.88 MB

08:42:25.527 -> ;echo:SDK: v4.4.5

08:42:25.527 -> ;echo:flash size: 4.00 MB

08:42:25.527 -> ;echo:size for update: 0 B

08:42:25.527 -> ;echo:FS type: LittleFS

08:42:25.560 -> ;echo:FS usage: 248.00 KB/896.00 KB

08:42:25.560 -> ;echo:baud: 115200

08:42:25.560 -> ;echo:sleep mode: none

08:42:25.560 -> ;echo:wifi: ON

08:42:25.560 -> ;echo:hostname: myesp

08:42:25.560 -> ;echo:HTTP port: 23

08:42:25.560 -> ;echo:Telnet port: 30000

I have set the http port with [ESP121] to 80 but still no connection

luc-github, 2023-11-15:
SDK: v4.4.5 this is not correct you must have SDK 4.4.4 
What esp32 arduino core are you using ? you are supposed to use 2.0.8 : https://esp3d.io/esp3d/v3.x/installation/arduino/index.html

luc-github, 2023-11-15:
also HTTP port: 23 is wrong it should be 80 that is why you cannot connect

juergmaier, 2023-11-15:
after setting the http port to 80 a reboot of the ESP shows now the same
webpage as I had in AP-mode.
The top section links point to:
Firmware: https://github.com/luc-github/ESP3D/tree/3.0
Interface: https://github.com/luc-github/ESP3D-WEBUI/tree/3.0
Help: https://github.com/luc-github/ESP3D/wiki

these sites are shown but I assume there is still something wrong with my
setup?

luc-github, 2023-11-15:
you use wrong core - a one I never tested

juergmaier, 2023-11-15:
my arduino board manager only allows to select:
 1.0.0 .. 1.0.6
2.0.10 and 2.0.11

and I do not know how to change SDK to version 4.4.4?

luc-github, 2023-11-15:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

juergmaier, 2023-11-15:
tried to execute the commands but get:

Saving to: ‘get-pip.py’

get-pip.py                 100%[========================================>]
  2.51M  --.-KB/s    in 0.1s

2023-11-15 09:42:44 (20.0 MB/s) - ‘get-pip.py’ saved [2632263/2632263]

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python
installation or OS distribution provider. You can override this, at the
risk of breaking your Python installation or OS, by passing
--break-system-packages.
hint: See PEP 668 for the detailed specification.
***@***.***:~$

luc-github, 2023-12-01:
your issue looks related to your platfom

closed 2023-12-23 by luc-github