luc-github/ESP3D #965: Add "Flip horizontally" for camera settings

DivingDuck, 2023-11-15:
**Feature request**

Is it maybe possible to add Flip horizontally to configuration.h some time in the future? I do add this often after an update manually and like to ask for adding the lines below:

```
/* Flip horizontally
* Flip camera horizontally
*/
//#define CAMERA_DEVICE_FLIP_HORIZONTALY
```
[image]

It would be also great having the controlls available in [Settings/Interface/Extra content Panel] or better(?) in [Dashboard/Camera Panel] as well so that it is possible to flip the view horizontal or vertical while running ESP3D-WebUI.

[image][image]

Not important but nice to have if needed.

Best regards,
DD

luc-github, 2023-11-15:
actually the define is already in the configurator :
[image]

I may missed to add the line in default configuration.h I will add it

About the extra buttons - I would like to avoid to add such button that will almost be never used used - on other hand you can add them by yourself in Macro panel using the [ESP170] command

luc-github, 2023-11-15:
Another possible solution if you want to play, is to do a little extension that has the display of camera,  do the refresh  and all buttons you need : https://esp3d.io/esp3d-webui/v3.x/documentation/api/extensions/index.html

DivingDuck, 2023-11-15:
Hey, new playground...
I will see how this works for me some time later today (hopefully) as I need to do some day work now. I will come back as soon had the time playing a bit with extensions (what I didn't before) :)

Again, thanks a lot

Btw, you can close the issue if you want. I updated my source file already and the rest I got managed with your hints. If not, I will come back and ask for help.

luc-github, 2023-11-15:
Sure 😺  have fun ^_^

commit 5fc3d4ae (2023-11-15): Fix #965 thanks @DivingDuck for raising it
closed 2023-11-15 by luc-github