luc-github/ESP3D #967: ESP32 Cam FS only 128kb and no CAM in extra content

xromansx, 2023-11-24:
Used the configurator to generate the files flash was succesefull after boot I have :
hip id: 49167
CPU Freq: 240Mhz
CPU Temp: 47.2C
free mem: 185.07 KB - PSRAM:4.00 MB
SDK: v4.4.3
flash size: 4.00 MB
size for update: 1.87 MB
FS type: LittleFS
FS usage: 8.00 KB/128.00 KB

FS type in configurator was set to FatFS

[image]

luc-github, 2023-11-24:
128kb is normal if you used configurator camera code use a lot so to allow web update I have set minimum partition as minimum spiffs 
about FATFs can you share your configuration.h?

luc-github, 2023-11-24:
also please give me the output of [ESP800]

xromansx, 2023-11-25:
[ESP800]
FW version:3.0.0.a216
FW target:unknown
FW ID:0
Setup:Enabled
SD connection:none
Serial protocol:Raw
Authentication:Disabled
Web Communication:Synchronous
Web Socket IP:192.168.0.1
Web Socket Port:81
Hostname:esp3d
WiFi mode:AP
Web update:Enabled
Flash File System:LittleFS
Host Path:/
Time:none
[configuration.zip](https://github.com/luc-github/ESP3D/files/13462758/configuration.zip)

luc-github, 2023-11-25:
no camera is defined in Fw for what I can see in ESP800

the configuration.h looks ok but it does not match the Firmware , it means you did not replaced the configuration.h of the repository with this one and compilation used existing one, you may copy it in wrong location

>And there is any posibility to move size for update into filesystem ?

sorry I do not undertand your question, FS and partition for update are different

xromansx, 2023-11-25:
I think the old config was cashed, after flashing I have:
hip id: 49167
CPU Freq: 240Mhz
CPU Temp: 48.3C
free mem: 128.28 KB - PSRAM:3.91 MB
SDK: v4.4.3
flash size: 4.00 MB
size for update: 1.87 MB
FS type: FAT
FS usage: 0 B/0 B
baud: 115200
sleep mode: none
wifi: ON
hostname: esp3d
HTTP port: 80
Telnet port: 23
camera name: ESP32 Cam(7)
ap: ON
mac: 10:52:1C:74:0F:C1
SSID: ESP3D

luc-github, 2023-11-25:
so with this one you ha fatfs and camera

xromansx, 2023-11-25:
Yes, thanks.  what about 128kb size, I saw in screenshots most people have ~4mb can I find somewhere documentation how this space can be extended and what for it can be used.

luc-github, 2023-11-25:
4MB is the total size of esp32 flash not the FS size that can be used

if you want to disable web update and ota    to increase FS size, you must create a partitions.csv file that fit your needs and use it instead of min_spiffs.csv

luc-github, 2023-11-25:
some ESP32 have 8MB or 16MB of flash so they can have 4MB of FS but not ESP32Cam

xromansx, 2023-11-25:
Thank you. Good to know so there is no posibility to print .gcode from internal FS only SD.

luc-github, 2023-11-25:
Using FS as SD is wrong good idea because using FS like this will make the flash to wear faster and so being unusable

the esp3d allow to stream gcode only for grbl and only in ping/pong mode for the moment and use esp32cam as cam and streaming host at once  may be over the capability of esp32cam and add latency in camera and streaming that may affect the streamed job

luc-github, 2023-11-25:
>And when you disable OTA and Web update from configurator it doesn't change the size of FS.

no, currently configurator is not smart enough for that, improvements are welcome if you want to add such capability

xromansx, 2023-11-25:
It would be interesting to have this option in configurator for now I increased the size of FS to 896kb removing ota1 and changing csv file.
My new file: 
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x300000,
spiffs,   data, spiffs,  0x310000,0xE0000,
coredump, data, coredump,0x3F0000,0x10000,

I usually print multiple copies of same print so mostly reading from FS. 
Thank you for your advice. 

After flash my config:
chip id: 49167
CPU Freq: 240Mhz
CPU Temp: 52.2C
free mem: 128.75 KB - PSRAM:3.91 MB
SDK: v4.4.3
flash size: 4.00 MB
size for update: 0 B
FS type: LittleFS
FS usage: 8.00 KB/896.00 KB
baud: 115200
sleep mode: none
wifi: ON
hostname: esp3d
HTTP port: 80
Telnet port: 23
camera name: ESP32 Cam(7)
ap: ON

luc-github, 2023-11-25:
ok so I guess issue.can be closed then

closed 2023-11-25 by luc-github