luc-github/ESP3D #973: [Question] [ESP3D v3] Hide notifications

dbuezas, 2023-12-03:
Is there any way to not display these notifications? 

[image]

It seems like these aren't hidden if the tab is not in the foreground for a while so they also accumulate.

dbuezas, 2023-12-03:
It looks like there is a bug on top of that. If I leave the tab in the background they accumulate and only the last one is removed:
[image]

luc-github, 2023-12-03:
what do you mean by `tab in background` ?

dbuezas, 2023-12-03:
For example if you have 2 tabs open in your browser, one is ESP3D and the other is youtube, but you are watching youtube so the ESP3d is "in the background", or inactive

closed 2023-12-03 by dbuezas