luc-github/ESP3D #986: [BUG] No IP Address on LCD/TFT of printer

firepower9966, 2024-01-01:
**Describe the bug**

No IP Address shown on LCD / TFT of printer in STA Mode. 
When in AP Mode it shows IP address and status messages.

Turn on or reset 3D Printer does not show IP address when connects in STA mode.

Display IP address on start up

**Screenshots**

**ESP3D Firmware:**

- ESP3D FW version:  FW: 2.1.3b0
- ESP3D-WebUI Version: UI: 2.1b77
- Wifi mode: STA
- Flash method: [PlatformIO]  Have used both, Arduino IDE 2.2.1 {to set FS size and full erase when got json error}
    If you build with Arduino:
    - Arduino ESP8266 package version: [e.g. 2.5.2] 
Generic ESP8266 

**Target Firmware:**
Marlin bugfix-2.1.x (Jan  1 2024 23:40:20)

**Board used (please complete the following information):**

- MCU:[ESP8266 ESP12S]
- Name:[MKS Wifi]
- Flash size: [4M: 2M/2M]

**Additional context**
I previously had IP address shown using ESP8266 D1 mini and ramps 1,4 Atmega2560 with LCD 12864 / LCD 2004
I am now using MKS Wifi with MKS Robin Nano V3,1 and TS35 V2,0 but no IP address, but everything else works fine
have tried with and without  #define DISABLE_CONNECTING_MSG
have tried changing Delay esp3d.h and still not IP, using 12s as it allows it to start and show firmware boot message in console of ESP3D.

[esp3d.zip](https://github.com/luc-github/ESP3D/files/13805635/esp3d.zip)

luc-github, 2024-01-02:
Does the IP appear on TFT if you do an `[ESP444]RESTART` in webui terminal ?

firepower9966, 2024-01-02:
Zip File contains my config.h and esp3d.h

luc-github, 2024-01-02:
[image]
no you put wrong file, you put config.cpp not .h

firepower9966, 2024-01-02:
here correct file. 
[esp3d.zip](https://github.com/luc-github/ESP3D/files/13808517/esp3d.zip)

luc-github, 2024-01-02:
Does the IP appear on TFT if you do an [ESP444]RESTART in webui terminal ?

firepower9966, 2024-01-02:
I currently printing , I will update later in about half an hour to complete print.

luc-github, 2024-01-02:
also does any command work from WEBUI to monitor / control printer ?

firepower9966, 2024-01-02:
[ESP444]RESTART  show disconnected on TFT
even when reconnects and i read sdcard or send command.
all commands work , gcode, mcode and [ESPcodes]
I can homee, set and monitor temp, fan, read and upload and print from sdcard/
everything seems to work ok , just no ip address on start up or reset

luc-github, 2024-01-02:
does `[ESP111]M117`  display the IP ?

firepower9966, 2024-01-02:
Only in Webui console
[ESP111]M117
M117192.168.1.11
I have already tried to do same as custom command sent from printer, but not display

luc-github, 2024-01-02:
what about : `M117 192.168.1.11` from webUI terminal

firepower9966, 2024-01-02:
Yes that displays 192.168.1.1
M117 works from console no problem.

firepower9966, 2024-01-02:
would adding space character before IP address allow M117 to show IP
[ESP111]M117
M117192.168.1.11 change to this  M117 192.168.1.1

luc-github, 2024-01-02:
yes you can add space to end
 `[ESP111]M117 `

firepower9966, 2024-01-02:
that shows in console 
[ESP111]M117
M117192.168.1.11

and no IP on TFT/LCD

luc-github, 2024-01-02:
I am checking  on it because the code is very old, I am adding possible solution now, 2.1 use HK wifi method not serial solution for emulating MKS WiFi module so I may missed that in 2.1

but I do not understand :
> ESP3D works fine and the MKS file transfer is great compared to the glacial slow transfer of standard serial wifi.

ESP3D 2.1 does not support MKS file transfert only 3.0 - I do not understand how can you have fast file transfert on 2.1 ?

firepower9966, 2024-01-02:
Its seems to work you have the option in config.h 
//MKS TFT WIFI support see Wiki for wiring
#define MKS_TFT_FEATURE

I can not get ESP3D 3.0 to work no wifi comms have tried twice,but ESP3D 2,0 work well

luc-github, 2024-01-02:
`MKS_TFT_FEATURE` just allow MKS TFT to get IP in settings and so make the communication active, it does not do anything on file upload

3.0 is a complete rewrite and has native support of MKS WIFI

luc-github, 2024-01-02:
I think it is fixed now I have tested with yourr config

firepower9966, 2024-01-02:
Maybe it just placebo , lol
I use USB stick for transfer as my PC is next to my printer.

firepower9966, 2024-01-02:
I will reflash firmware with MKS TFT WIFI  disabled. maybe thats the issue
//MKS TFT WIFI support see Wiki for wiring
//#define MKS_TFT_FEATURE

luc-github, 2024-01-02:
if you do this the TFT may think there is not WIFI module, just download the FW on git I have already pushed the fix
https://github.com/luc-github/ESP3D/commit/460a145a91588f7451fe7016b8db6c0a67758b4c

firepower9966, 2024-01-02:
Thats it now see IP 
now shows 
Wifi ssid 
disconnect 
IP

firepower9966, 2024-01-02:
MY wifi in on mainboard 
I will update to latest 2.1 ?

luc-github, 2024-01-02:
yes or change the code I have linked to esp3d.cpp

firepower9966, 2024-01-02:
Working OK
see IP on TFT

luc-github, 2024-01-02:
cool thank you for reporting it 😺

luc-github, 2024-01-02:
I close issue then

closed 2024-01-02 by luc-github

firepower9966, 2024-01-02:
yes thank you. 
Would having 
//MKS TFT WIFI support see Wiki for wiring
#define MKS_TFT_FEATURE

also cause my issue of no wifi coms when trying to use ESP3D 3.0 ?

luc-github, 2024-01-02:
for more info you can check here : https://esp3d.io/esp3d/v3.x/index.html