luc-github/ESP3D #988: [BUG]Error compiling on ESP170.cpp

Engineer2Designer, 2024-01-04:
**Describe the bug**

**To Reproduce**

1. Go to compiling 
2. on  'esp3d/src/core/commands/ESP170.cpp: In lambda function:'
3. See error

compiling log

Processing esp32cam (platform: espressif32@6.0.1; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Check if need to add some library to path
['C:\\Users\\E2D\\Documents\\PlatformIO\\Projects\\ESP3D-e2d\\libraries', '$PROJECT_LIBDEPS_DIR\\$PIOENV', 'C:\\Users\\E2D\\.platformio\\lib']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.0.1) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Converting esp3d.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFi @ 2.0.0
|-- EEPROM @ 2.0.0
|-- Preferences @ 2.0.0
|-- WebServer @ 2.0.0
|-- BluetoothSerial @ 2.0.0
|-- Ticker @ 2.0.0
|-- Wire @ 2.0.0
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0
|-- TFT_eSPI @ 2.4.72
|-- FS @ 2.0.0
|-- SD @ 2.0.0
|-- SPI @ 2.0.0
|-- FFat @ 2.0.0
|-- LittleFS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- SD_MMC @ 2.0.0
|-- ESP32SSDP @ 1.2.0
|-- Update @ 2.0.0
|-- ESP8266-Arduino-Lua @ 0.0.30
|-- ESPmDNS @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- DNSServer @ 2.0.0
|-- HTTPClient @ 2.0.0
|-- WiFiClientSecure @ 2.0.0
|-- BMx280MI @ 1.2.0
|-- DHT sensor library for ESPx @ 1.0.6
|-- WebSockets @ 2.3.6
|-- Ethernet @ 2.0.0
Building in release mode
Compiling .pioenvs\esp32cam\src\esp3d.ino.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP0.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP100.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP101.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP102.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP103.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP104.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP105.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP106.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP107.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP108.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP110.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP111.cpp.o
Compiling .pioenvs\esp32cam\src\src\core\commands\ESP112.cpp.o
[... 187 lines cut ...]
myconfig.h

[myconfig.h.TXT](https://github.com/luc-github/ESP3D/files/13824953/myconfig.h.TXT)

**Screenshots**

**ESP3D Firmware:**

- ESP3D FW version:
- ESP3D-WebUI Version:
- Wifi mode: STA
- Flash method: PlatformIO  

**Target Firmware:**

- Name: Marlin
- Version 2.1.2

**Board used (please complete the following information):**

- MCU: ESP32
- Name:[e.g. ESP32-CAM
- Flash size: 4M

**Additional context**

luc-github, 2024-01-04:
seems there are some typo, orz...
I am out now I will check and fix when back home in few hours

luc-github, 2024-01-04:
It is now fixed in latest git with https://github.com/luc-github/ESP3D/commit/00579614975d7cac20e63346eb41befd94a037ee

Engineer2Designer, 2024-01-05:
@luc-github has repaired it, thank you.

commit 00579614 (2024-01-04): Fix compilation issue due to refactoring typo error thanks @Engineer2Designer for raising it

Close #988
closed 2024-01-05 by Engineer2Designer