luc-github/ESP3D #991: [FEATURE REQUEST]<Heltec-rastreador inalámbrico WiFi/LoRa/BLE/GNSS(L5)>

vicen1206, 2024-01-10:
Hola, compre esta placa de desarrollo: Heltec-rastreador inalámbrico WiFi/LoRa/BLE/GNSS(L5),
Me gustaria usarla como un Traker de lora , como los TTgo, pero esta tiene pantalla TFT y el programa solo tiene para pantalla OLED, cabe la posibilidad de habilita la TFT y como se aria si fuera posibre, adjunto manual de la placa,
[Wireless Tracker 1.1.pdf](https://github.com/luc-github/ESP3D/files/13887047/Wireless.Tracker.1.1.pdf)
Gracias de ante mano

luc-github, 2024-01-10:
esp3d already support esp32s3 and some oled
what is your issue using esp3d on it?

vicen1206, 2024-01-10:
Gracias por contestar tan rápido, hago la programación con el Visual studio y todo perfecto pero no se visualiza nada en la pantalla,

luc-github, 2024-01-10:
do you have heltec sample code to check hardware is ok?
also sample code would allow to check your pinout is correct

vicen1206, 2024-01-10:
si, una de aqui, https://flasher.meshtastic.org/ y funciona a la perfeccion

vicen1206, 2024-01-10:
Mi intención es usarlo con esta programación
[LoRa_APRS_Tracker-main (1).zip](https://github.com/luc-github/ESP3D/files/13889527/LoRa_APRS_Tracker-main.1.zip)

luc-github, 2024-01-10:
what screen this board is using?

vicen1206, 2024-01-10:
es una TFT 0.96" 80x160

luc-github, 2024-01-10:
what chipset?

vicen1206, 2024-01-10:
[Wireless Tracker 1.1.pdf](https://github.com/luc-github/ESP3D/files/13891126/Wireless.Tracker.1.1.pdf)
[image]
[image]

luc-github, 2024-01-10:
yes but I do not see what drivers this screen need in the doc neither what chipset it use

vicen1206, 2024-01-10:
no se si esto te sirve 
[image]

vicen1206, 2024-01-10:
[image]

luc-github, 2024-01-10:
it looks like from the github it is st7735

so you need to do change accordingly as this chip is supported by TFT_eSPI lib

I do not have this board but look like overkill if you do not use its gps / lora features as esp3d does not support them

vicen1206, 2024-01-10:
Entonces puedo usarla con lora? si es asi , como lo hago?  aun que sea de I-Gate en vez de Traker

luc-github, 2024-01-10:
esp3d is for 3D printer and cnc it does not make sense to support lora or gps

I do not understand your question

vicen1206, 2024-01-10:
entonces no puedo instalar LoRa_APRS_Tracker-main o LoRa_APRS_iGate-master a esta placa?

luc-github, 2024-01-10:
this is not lora github why asking that?

vicen1206, 2024-01-10:
ok, muchas gracias por todo

luc-github, 2024-01-11:
I close issue then

closed 2024-01-11 by luc-github