luc-github/ESP3D #996: Got error "connection with board is lost" and  "websocket connection error"

dialdhruv, 2024-02-04:
Bundrate is 250000
ESP3D-WEBUI-3.0.0-alpha3
ESP3D-3.0.0-alpha3

using Webui \ESP3D-WEBUI-3.0.0-alpha3\server\Printer3D\Marlin\Flash

When I load the page it works properly for 2-3 seconds then after it shows the below error.

[image]

[image]

**I have tried with firmware 2.X and it is working fine for me. No any issue was found with 2.X firmware version**

luc-github, 2024-02-05:
the webui and fw are not sync , which mean  one of then is not alpha3
please use latest git version for both it should be ok

dialdhruv, 2024-02-05:
> Thank your for submiting, please be sure you followed template or your issue may be dismissed. if you deleted the template it is [here](https://github.com/luc-github/ESP3D/blob/2.1.x/.github/ISSUE_TEMPLATE/bug-report---problem.md)

I did not understand template. Are you talking about webui?
Webui and firmware version both 3.0.0-alpha3

dialdhruv, 2024-02-05:
> the webui and fw are not sync , which mean one of then is not alpha3 please use latest git version for both it should be ok

I have download both from github release and both version is template 3.0.0-alpha3

luc-github, 2024-02-05:
I mentionned git version - not release 

[image]

luc-github, 2024-02-07:
I close issue as cannot duplicate on git version

closed 2024-02-07 by luc-github

luc-github, 2024-02-07:
I have also tested with https://github.com/luc-github/ESP3D/tree/v3.0.0-alpha3 and https://github.com/luc-github/ESP3D-WEBUI/tree/v3.0.0-alpha3 and cannot duplicate