luc-github/ESP3D #1000: Pinter’s memory card is not showing in esp3d 3.x

flyroboofficial, 2024-02-09:
Printer’s Sd card is not showing in esp3d version 3.x
While it is showing in esp3d 2.xx version.

rondlh, 2024-02-09:
> Printer’s Sd card is not showing in esp3d version 3.x While it is showing in esp3d 2.xx version.

What do you mean? Is the button for the SD card not present in the WebUI "files panel", or is the contents of the SD card is not shown when you press the "TFT SD" button?

flyroboofficial, 2024-02-09:
I can able to upload files to esp32 module only. Sd card storage is not showing. Only esp32 storage is showing.

luc-github, 2024-02-10:
Did you enable SD in settings ?
[image]

flyroboofficial, 2024-02-10:
yes, I have enabled the setting. after updating latest firmware 3.0 SD card is showing. Now problem is I can not able to upload file to the SD card remotely.
No option to upload file.
[image]

luc-github, 2024-02-10:
in ESP3D 3.0 here the supported configurations : https://esp3d.io/esp3d/v3.x/documentation/sdtransfer/index.html
the M28/M29 protocol is no more supported because too slow / suject to many error and actually very very few people use it - so it doe s not worth it
Direct SDC is supported but streaming is not yet so currently only MKS configurations and system with hardware SD sharing are supported

luc-github, 2024-02-10:
Are you in supported cases ?

flyroboofficial, 2024-02-10:
No, I am using the RAPM 1.6 board.

[https://www.flyrobo.in/ramps-1.6-r6-3d-printer-control-board](https://www.flyrobo.in/ramps-1.6-r6-3d-printer-control-board)

luc-github, 2024-02-10:
then you must add an SDCard reader with sharing mechanism like panucatt wifi backpack

flyroboofficial, 2024-02-10:
ok, I Understend.
SD card upload is working in ESP3D 2.x version. So why this option has beed removed in updated version? i think it was good option.

luc-github, 2024-02-10:
I have just explained in previous comment : https://github.com/luc-github/ESP3D/issues/1000#issuecomment-1936922949
Many people complained it is slow and cannot be use because of this even I explained it how it is designed : https://github.com/luc-github/ESP3D/discussions/535
So instead of implementing something people are not happy with and complain about it, I use my time on others features

flyroboofficial, 2024-02-10:
Ok thank you

closed 2024-02-10 by flyroboofficial