luc-github/ESP3D #1001: [FEATURE REQUEST] Ability to schedule starting a print at a given date/time

josedpedroso, 2024-02-17:
It'd be nice to be able to schedule printing a file already in the card (or sending a custom g-code command to the printer) at a specified date and time. Not sure how feasible that is, especially if ESP3D doesn't already have a global date/time, but as an alternative, being able to schedule the print or command to happen after some minutes, days, or hours, would still be useful. It'd also be nice to be able to set the schedule as one-shot or repeating.

luc-github, 2024-02-17:
what is the purpose of scheduling a jog ? what is the user case?

josedpedroso, 2024-02-18:
My printer is a bit noisy and I have it at my parents', so I'd like to
avoid running it at night. However, I'm often not at home in the morning to
start a job, so I may not be able to start long prints until I actually
spend the night there. Scheduling would solve this for me, without having
to setup remote access and without having to get up in the morning.

luc-github, 2024-02-18:
so what you need is just starting time and not a repeating 
In that case it is safer to use https://fr.aliexpress.com/item/1005004251720806.html because if your esp does not have internet and time may not be set 
Just put you gcode file as init command in ESP3D or even in Printer Firmware init file

josedpedroso, 2024-02-18:
As I said in the first message, even just scheduling for after some
minutes/hours/days, without an absolute date/time, would still be helpful.

luc-github, 2024-02-18:
My todo list is already full of helpful feature requests  so I need to sort them by "is it useful ?" "Does manyl people will use it ?" "Is there an existing solution ?"

So to be honest currently this FR would the last one

josedpedroso, 2024-02-18:
I'm ok with that.

kleffa, 2024-06-01:
Launch macros on determinate time sounds very interesting, by now using macros or comands like "G4 S3600 ;" makes a solution to pause at some time, but have lit of limitations, some printers have lights, buzzers and noises that can easily activate/deactivate using macros on time, by now ESP3D has NTP so sontime should not be a gteat problem

luc-github, 2024-06-01:
sure, PR is welcome

closed 2024-08-09 by luc-github