luc-github/ESP3D #1002: Request for SD card access SMOOTHIE

madebywitteveen, 2024-02-28:
Hello all, how are you?

Trying to get WIFI access working to my WAZER waterjet with SMOOTHIE board.

[image]

The machine is responding to the manual x/y movements, that is the good part!
How can I tell whether the ESP3D in combination with this SMOOTHIE is capable of reading the SD CARD at all?
The SD card field on the dashboard stays empty.

When trying to upload a file this error pops up:
[image]

If possible I would really like to have this fixed.

thanks

Henk

luc-github, 2024-02-28:
reading SD card and upload sd card are 2 different things 
can you list the content of SD card in webui ?

madebywitteveen, 2024-02-29:
Hello Luc, thank you for answering;

 

The contents of the card red from a normal card reader in my pc:

 

Here it is from the ESP3D:

 

 

Can I open some more detailed debugging logfile or see the SD card communication somewhere?

 

Thanks

 

Henk

luc-github, 2024-02-29:
I do not see anything - did you uploaded pictures ?

madebywitteveen, 2024-02-29:
Hello Luc,No pictures,  just gcode files.The SD card is a kingston 32GB version formatted as 256mb.Can I see the communication between the card and the wemos somewhere to give you some debugging information?ThanksH

luc-github, 2024-02-29:
the pictures was about your answer that I cannot see , I have asked you : can you list the content of SD card in webui ?
and you answered 

>Hello Luc, thank you for answering;
>
>The contents of the card red from a normal card reader in my pc:
>
>
>
>Here it is from the ESP3D:
>
>
>Can I open some more detailed debugging logfile or see the SD card communication somewhere?
>
>Thanks
>Henk

So I ask again :  can you list the content of SD card in webui ?

madebywitteveen, 2024-02-29:
Hi Luc,No I can't list the files in the webui Thankyou

luc-github, 2024-02-29:
what about the output of following commands  :
1 - M2.0
2 - ls /
3-  ls /sd
4 ls /ext

madebywitteveen, 2024-02-29:
Hello Luc,

 

unfortunately that doesn’t do anything and or show anything.

 

Thank you, H

luc-github, 2024-02-29:
Does `M114` display machine position ?

madebywitteveen, 2024-03-02:
Hello How are you ?
No unfortunately it gives back an error:

[image]

For the error to appear it takes 10 seconds or so.

Can you provide me with a list off all the m-codes? 
Thank you

luc-github, 2024-03-02:
what is the `[ESP410]` output ?
looks like you have communication issue

madebywitteveen, 2024-03-02:
Hello Luc,

the result of ESP420 is:

[ESP420]plain
Chip ID: 12599193
CPU Frequency: 160Mhz
Free memory: 33.00 KB
SDK: 2.2.2-dev(38a443e)
Flash Size: 4.00 MB
Available Size for update: 506.05 KB(Not enough)
Available Size for SPIFFS: 1.86 MB
Baud rate: 115200
Sleep mode: None
Channel: 2
Phy Mode: 11g
Web port: 80
Data port: 8888
Hostname: SMOOTHIE
Active Mode: STA (A8:48:FA:C0:3F:99)
Connected to: AQANQN
Signal: 70%
IP Mode: DHCP
IP: 192.168.1.79
Gateway: 192.168.1.1
Mask: 255.255.255.0
DNS: 213.46.228.196
Disabled Mode: AP (AA:48:FA:C0:3F:99)
Captive portal: Enabled
SSDP: Enabled
NetBios: Disabled
mDNS: Enabled
Web Update: Enabled
Pin Recovery: Disabled
Authentication: Disabled
Target Firmware: Smoothieware
M117 output: Enabled
Notifications: Disabled
Serial output: Enabled
Web socket  output: Enabled
TCP output: Enabled
FW version: 2.1.3b0 ESP8266/8586

Thank you

luc-github, 2024-03-02:
signal is low but you seems getting esp3d response - but in all case because you mention the command can move the printer,  there is no response from printer which means you have issue with the rx pin

madebywitteveen, 2024-03-02:
You are right, I did fix that.
A lot of information is scrolling by.
Disabling the 'autoscroll' option doesn't help.
What I did find out it is that code M2.0 and M118 gives response back in red "command not supported"
The SD-Card section stays empty
Does this give you insight what you need ?

Best regards

luc-github, 2024-03-02:
M2.0 was a typo
I need 
M20 
ls /sd
ls /
ls /ext

madebywitteveen, 2024-03-02:
Hi,

The information is scrolling too fast, can't read it but I see something red flashing after typing M20.
I guess that means 'command not supported' 

thanks
H

luc-github, 2024-03-02:
scrolling too fast ?

madebywitteveen, 2024-03-02:
https://github.com/luc-github/ESP3D/assets/63544034/423baa41-8152-41c9-967a-80ab095898bd

luc-github, 2024-03-02:
That is not standard output 
not sure what it is

luc-github, 2024-03-02:
also it is M20 not m20

madebywitteveen, 2024-03-02:
Thanks, perhaps in the communication between the WEMOS D1 mini and the SMOOTHIE things are overloaded or the flowcontrol isn't properly working.
Had my scope (with UART decoding) hooked-up and some frames are having under and over length.
or the wires are picking-up to much noise or something

going to watch F1 now, you ?

H

luc-github, 2024-03-02:
not really the format 
o:wrc 47 47 0
:ack  81 
is pretty clear and it is not gcode reponse - may be you have custom firmware

madebywitteveen, 2024-03-02:
That could be the case, I have no clue what firmware is inside.
Are you perhaps willing to develop this ?
Say you help me out finding out how to get this working; what do you need from me for debugging purposes?
Willing to buy you a coffee !

DivingDuck, 2024-03-02:
Hi,
this looks like the uart baud rate from smoothie board is different to what is expected from esp3d. Check the config file (-->## System configuration -->uart0.baud_rate or -->uart1.baud_rate) to see what is set up there and put that in the settings of esp3d (-->Settings -->Features -->System)

madebywitteveen, 2024-03-02:
@DivingDuck Thanks for the heads up but I am afraid that I need a bit more information from you.
Which config file ? 'config.h' ? 
And which section do I have to check? 
You are referring to version V3, mine is currently V2. Should I upgrade ?

Much appreciated

DivingDuck, 2024-03-02:
The config file from smoothieware. Usually it is called config or config.txt and can be found at your sd-card in the smoothie controller. 
Section is:
```
## System configuration
# Serial communications configuration ( baud rate defaults to 9600 if undefined )
# For communication over the UART port, *not* the USB/Serial port

``` 

I'm using version 3 of esp3d in combination with ESP32-CAM, but there should be a similar config for version 2. Guess this should be DEFAULT_BAUD_RATE. Default rate is usually 115200 in esp3d. Check that both, smoothie config and esp3d, use the same rate.

Should you upgrade? 
This is up to you. I would maybe first establish the version you are just testing and then upgrade if you want to check version 3.

luc-github, 2024-03-02:
> That could be the case, I have no clue what firmware is inside. Are you perhaps willing to develop this ? Say you help me out finding out how to get this working; what do you need from me for debugging purposes? Willing to buy you a coffee !

Develop a Full FW without even having the system for a coffee - you are too generous - I skip ^_^

madebywitteveen, 2024-03-03:
@DivingDuck The firmware is partly readable and it looks like that the baudrate is the same. (115200)

[FIRMWARE.txt](https://github.com/luc-github/ESP3D/files/14472732/FIRMWARE.txt)

thanks

madebywitteveen, 2024-03-03:
> Develop a Full FW without even having the system for a coffee - you are too generous - I skip ^_^

I'm sorry, I wasn't clear; not developing a full FW, only trying to find out where my SMOOTHIE is listening to and which M codes can be used and which not.
The FW is in the post below

luc-github, 2024-03-03:
sorry I do not have your system and I am not magician I cannot debug a system blindly
for available gcode just read the documentation https://smoothieware.github.io/Webif-pack/documentation/web/html/index.html

luc-github, 2024-03-03:
and checking https://support.wazer.com/miscellaneous-downloads/firmware-and-config-files the config.txt show that the FW use customized tool code and custom code removing several parameters - so doing reverse engineering on it is challenging, even more without a system

DivingDuck, 2024-03-03:
@madebywitteveen,
sorry for my late answer. I guess the attached file firmware.txt is the bin file firmware.bin. and missing is the config.txt file.
The firmware file include most of the configuration and is not part of the config file. So most of the configuration is not visible for the user. I know that smoothieware have a option to build the firmware that way. Bad for the user...

Anyway, with the link from @luc-github I saw that the original config file do not include the most of configuration information (like baud rate or their customized M-codes). But looking to the bin file I can identify some of the config information.
uart0.baud_rate is set to 115200. 

Some M-codes I saw in the firmware (not complete):
```
switch.low_pressure_pump.input_on_command    M31
switch.low_pressure_pump.input_off_command   M33
switch.vibration_motor.input_on_command      M46
switch.pump_box_ssr.input_off_command        M43
switch.high_pressure_valve.input_on_command  M3
switch.high_pressure_valve.input_off_command M5
switch.abrasive_valve.input_on_command       M8
switch.abrasive_valve.input_off_command      M9
switch.dump_valve.input_on_command           M44
switch.dump_valve.input_off_command          M45

```

Guess, it is better to ask the manufacturer for a complete list of all G- and M-codes or maybe you can find them in the documentation to your device.

Maybe also useful for research:
https://forum.lightburnsoftware.com/t/support-request-for-wazer-waterjet/103922
You can use web.archive.org since smoothieware home page isn't any longer available to find some more information regarding smoothieware:
https://web.archive.org/web/20231001110406/http://smoothieware.org/
https://web.archive.org/web/20230929125715/http://smoothieware.org/supported-g-codes
https://web.archive.org/web/20230610144407/http://smoothieware.org/console-commands

Regarding the errors in communication I can only speculate since the baud rate seems to be on both 115200 baud. Maybe you need to use a shielded cable between the controller board and the ESP. I think it is possible that there is some electromagnetic interference in this machine and this is usually not good for serial communication with two wires.

Guess, you will have a lot of work in front of you using esp3d with your water jet machine.

luc-github, 2024-03-04:
So for what I can see there is no support for this custom FW so move issue as discussion as it is not a bug but limitation