luc-github/ESP3D #1009: Wifi not working on ESP32C3 Super mini (and possibly other C3 boards)

IntEA31, 2024-03-07:
Compiled version 3.0 from sources using platformio. After uploading into board, no ESP3D AP is visible and only posibility to communicate with living ESP3D is via serial line. No obvious reason based on available logs. 

After searching through various internet sources, it seems that some ESP32 C3 boards seems to have hw defficiency within power supply design preventing the chip communicating on higher (and possible default) Tx power levels. 

Based on this, wificonfig.cpp needs to be updated (patch attached) to explicitly call WiFi.setTxPower() in StartSTA() StartAP() methods and (possibly not needed) begin() methods. Default power level set to 15dBm (defined by TX_POWER macro, available also from config.h). Confirmed working at least with my unnamed board.

The same problem and principal solution also apply to v2.1 code base.

[ESP32C3-txpower.patch](https://github.com/luc-github/ESP3D/files/14525864/ESP32C3-txpower.patch)

luc-github, 2024-03-07:
I have several C3 boards from different suppliers and no issue so far but I do not have ESP32-C3 supermini board
can you link the board supplier and also the site that suggest the fix ?

IntEA31, 2024-03-08:
Hello Luc.
Thanks for the feedback. Yes, this is probably related to hw bug on specific boards - see comments on https://github.com/espressif/arduino-esp32/issues/6430 where's Wemos Lolin C3 board mentioned. Mine board is the TenStar version of ESP32C3 from Aliexpress, eg. https://www.aliexpress.com/item/1005006211660074.html Identifiable by the buttons directly behind usbc connector. Pinout and schematic could be found on https://www.nologo.tech/product/esp32/esp32C3SuperMini.html

I think at least comment/note in config.h about TxPower lowering could save other some time in searching when using such problematic boards.

luc-github, 2024-03-08:
config.h is for esp3d 2.1.x which is not supposed to work with C3 because not tested

ESP3D V3 support ESP32-C3 so the `WiFi.setTxPower(XXX);` should be there

But this patch should only apply when needed , not on all ESP32 boards, 

Also value to apply seems to be WIFI_POWER_15dBm, WIFI_POWER_8_5dBm,  WIFI_POWER_5dBm depending of the board

I need to think about it , how to add it to configurator and notes

luc-github, 2024-03-08:
looking at https://github.com/arendst/Tasmota/discussions/15443#discussioncomment-2755063
the patch seems allowing to boot but usage is bad , the issue is a faulty component 
@IntEA31 how is you board responsiveness after patch ?

IntEA31, 2024-03-10:
You're right, mixing v2.1 and v3.0 config, I meant configuration.h file in one of posts above.

I was affraid about stability too, so I let sit my board over weekend connected and ran monitoring via ping and also made several prints during that time. I didn't saw any problem with connection dropouts to esp3d webui or dropped communication with the printer. That all with 20cm dupont "sparrow's nest" suboptimal connection. 

In my case 15dBm level was the highest safe level when the board was visible on network, so I went one step down just to be sure.  I suspect that the small onboard 3V3 stabilizer could not withstand the peak current request on higher power levels, but this is just speculacy. As far as I can tell the board is 100% stable when used with lower Tx power setup.

For clean implementation the TX_POWER value could be taken from configuration.h (or my_config.h if present) and then the call for WiFi.setTxPower could be made if defined. Another patch proposal is here - [ESP32C3-txpower2.patch](https://github.com/luc-github/ESP3D/files/14551160/ESP32C3-txpower2.patch)

luc-github, 2024-03-16:
I did this but cannot reproduce issue even without patch on my C3 super mini
https://github.com/luc-github/ESP3D/tree/C3-Branch-patch

[image]

[image]

Let me know if ok - I have added the delay based on others people test

IntEA31, 2024-03-18:
Hello Luc.
You were probably luckier or I exhausted my fortune obtaining wrong board batch - all my 3 devices has the same symptoms. I tried to compile your extra branch and can confirm it is working in both - STA and AP modes. This way everything is cleanly prepared for anyone with such problematic boards like me. This time I didn't run any deep tests as the logic of your patch is the same, just wanted to confirm you that this code could be included in mainstream branch.
Thanks for your effort in software fix of hw issue.

luc-github, 2024-03-20:
PR pushed - I will add the board in https://esp3d.io/esp3d/v3.x/hardware/esp_boards/esp32-c3/index.html and a comment to explain the define to use
I do not plan to add it in configurator to avoid confusion, people having issue will have to refer to https://esp3d.io/esp3d/v3.x/hardware/esp_boards/esp32-c3/index.html

closed 2024-03-20 by luc-github

luc-github, 2024-03-20:
page added: https://esp3d.io/esp3d/v3.x/hardware/esp_boards/esp32-c3/esp-c3-super-mini/index.html