luc-github/ESP3D #1012: [Question] Some clarifications

ilko-k, 2024-03-22:
First question, is this project still maintained ? No release since more than a year.

Second, some things differ bewteen the master branch here on git and the online configurator regarding the _platformio.ini_ file.

• **board_build.filesystem = littlefs** → not present in master
• **lib_ignore = ESP32SSDP** → is ESP32SSPD in master ( **D** and **P** are inverted ) 

Which one is up to date ?

luc-github, 2024-03-22:
Reading the readme should answer all your questions:
[image]
No mention you can check the commit history to see even no release there was a fix 2 month ago

>board_build.filesystem = littlefs → not present in master

Well 2.1 only support SPIFFS ...

>lib_ignore = ESP32SSDP → is ESP32SSPD 

Wow orz...  I missed that - Thank You 2.1 and 3.0 updated

ilko-k, 2024-03-22:
Allright thanks!