luc-github/ESP3D #1015: Web socket connection error message when using port 80  mapping on ESP32 network repeater

martiinek, 2024-03-24:
Hello, my ESP32CAM, which runs the ESP3D software, is connected to an ESP32 wifi repeater to increase signal strength and the repeater is connected to a router. I use this [https://github.com/martin-ger/esp32_nat_router] (firmware) on the ESP32 repeater and it works great. To access the web servers located on the devices connected to the ESP32 repeater from the main router, I need to map the web server port (http port 80). After I do this and visit the esp3D dashboard(http://esp3D_ip:port), within seconds I get "websocket connection error" followed by "connection error" "connection to board lost" and a reconnect button. After pressing the button it's still the same problem. Is there any fix?

luc-github, 2024-03-24:
any fix of what ? sorry are you asking a fix for ESP32 repeater here ?

martiinek, 2024-03-24:
I don't understand why I keep getting this error message while the webserver port is being mapped.

luc-github, 2024-03-24:
because websocket use another port may be https://esp3d.io/esp3d/v2.x/documentation/api/index.html#web-socket

luc-github, 2024-03-24:
I have update the doc for 3.x https://esp3d.io/esp3d/v3.x/documentation/api/websockets/index.html#terminal-websocket