luc-github/ESP3D #1018: [BUG] Compiling 3.0 in PIO env:esp8266dev fails

lenny-chem-dat, 2024-03-25:
**Describe the bug**

When trying to compile ESP3D 3.0 with PIO for any esp8266 environment with an SD card, the compiler fails with the following error messages:
esp3d\src\modules\filesystem\sd\sd_native_esp8266.cpp:162:3: error: 'FSInfo64' was not declared in this scope; did you mean 'fs::FSInfo64'?
esp3d\src\modules\filesystem\sd\sd_native_esp8266.cpp:165:22: error: 'info' was not declared in this scope

**To Reproduce**

1. In configuration.h, set #define SD_DEVICE ESP_SD_NATIVE
2. Select environment env:esp8266dev or env:esp01s_160mhz
3. Compile
4. See error

**Expected behavior**

Complete compilation

**ESP3D Firmware:**

- ESP3D FW version: 3.0 (including Commits on Mar 3, 2024)
- ESP3D-WebUI Version: N/A
- Wifi mode: N/A
- Flash method: PlatformIO
- 
- Any change done in FW: see attached configuration.h

**Additional context**

This seems to be triggered by the section uint64_t ESP_SD::usedBytes, in file esp3d\src\modules\filesystem\sd\sd_native_esp8266.cpp which is different from the code in the file esp3d\src\modules\filesystem\sd\sd_native_esp32.ccp. The latter is working fine, but esp8266 seems to handle the request differently. Also, switching to ESP_SDFAT2 doesn't compile due to errors in the esp3d\src\modules\filesystem\sd\sd_sdfat2_esp8266.cpp

[configuration.zip](https://github.com/luc-github/ESP3D/files/14740548/configuration.zip)

luc-github, 2024-03-25:
weird I did not do anything on SD for a long time  - I have duplicated it

luc-github, 2024-03-25:
Indeed SD support is broken for a long time orz I did several update on ESP32 and forget esp8266.. 
I am on it now

luc-github, 2024-03-25:
Fixed in version 232
NATIVE and FAT32 now properly compile and work - thank you for raising it 

[image]

closed 2024-03-25 by luc-github

lenny-chem-dat, 2024-03-25:
Great work, thank you! I’ll check it later today