luc-github/ESP3D #1020: [BUG]<Enter comprehensive title>

kleffa, 2024-05-05:
**Describe the bug**

C:\Users\kleff\Desktop\ESP3D-3.0-alpha2\esp3d\src\core\hal.cpp:30:10: fatal error: soc/rtc_wdt.h: No such file or directory
   30 | #include <soc/rtc_wdt.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: soc/rtc_wdt.h: No such file or directory

**To Reproduce**

download relase from this page ESP3D-3.0-alpha2
open in arduino IDE

notify force to rename littlefs_esp8266_filesystem .cppo to littlefs_esp8266_filesystem.cppo (has a space)

**Expected behavior**

flashing the esp32

**Screenshots**

[image]

**ESP3D Firmware:**

- ESP3D FW version: 3.0 beta 1 (also tried with other version, same troubles)
- Flash method: Arduino Ide 
- Any change done in FW: nothing

closed 2024-05-05 by luc-github

kleffa, 2024-05-06:
This still be a problem in Arduino IDE 2

luc-github, 2024-05-06:
https://esp3d.io/esp3d/v2.x/installation/arduino/

luc-github, 2024-05-06:
https://esp3d.io/esp3d/v3.x/installation/arduino/