luc-github/ESP3D #1021: [FEATURE REQUEST]Extra SD with M200 and M200L

kleffa, 2024-05-06:
I see the SD root of my 3d printer with M200 and works!
 but if I try use M200L for entire name of files, M23 only works with M200 
so I have to chose see entire name of files or comand printer to work blinded

**Describe the solution you'd like**
I think just asking printer M200 and M200L then show user M200L, but play button with the M200 version, printer send same list so compare both lists should be easy

**Describe alternatives you've considered**
For now I list M200L so remember position, so change to M200
**Additional context**

this work to play
[image]
 this is better for users, but doesnt play
[image]

luc-github, 2024-05-06:
M200 ? https://marlinfw.org/docs/gcode/M200.html ?

Why using M20 but display M20 L ?

luc-github, 2024-05-06:
no if you enable long name in Marlin `M20 L`
then `M23 long_name.gcode` is supposed to work and I tested before ot was

kleffa, 2024-05-06:
in marlin documentation, only shows short name 
https://marlinfw.org/docs/gcode/M023.html
also a closed issue of this long ago in marlin 
https://github.com/MarlinFirmware/Marlin/issues/3079 

so basically it was no a feature and marlin have no intention to work with long names.
maybe a simple solution should be add a button next to play to ask m33 of a short name file

luc-github, 2024-05-06:
no Marlin is now supposed to support long name, there  were several PR for this

kleffa, 2024-05-06:
I can´t find that M23 doesn´t work with long filename... maybe in futrure versions of marlin, but still considering almost easy implement play button for this kind of lines 
CE46F2~1.GCO 4853684 CE3e08-headband-side-right-5h42m-120g.gcode to
CE3e08-headband-side-right-5h42m-120g.gcode [M23 CE46F2~1.GCO ; M24]

luc-github, 2024-05-06:
you did ot search a lot here one of the PR https://github.com/MarlinFirmware/Marlin/pull/25540

as wrote it is.Marlin bug, I do not plan to workaround because it will not fix ESP3DLib which use shared SD

kleffa, 2024-05-06:
maybe just a change on the play button on the webUI, the #define UTF_FILENAME_SUPPORT is not suported universally, requires more memory and that stuff

luc-github, 2024-05-06:
UTF is not long name is related to unicode and multibyte chars

closed 2024-05-06 by luc-github

kleffa, 2024-05-06:
NOP doesn´t worked at all