luc-github/ESP3D #1030: [Question] Communication problem between ESP3D and the printer

KristiyanHristov, 2024-06-05:
I have a problem with communication between ESP3D and the printer. UI claims that command are sent but data is not received back. ("ESP3D says: command forwarded") 

Can you help me with debugging this issue?

Thank you!

**What I've tried:**
1. connecting RX/TX to the CH340 chip directly
2. connecting RX/TX to the screen output
3. connecting RX/TX to the screen output (in place of the screen)
4. using both UART0 and UART1 output ports on the ESP32 (only pins TXD, RXD seem to do something)
5. swapping RX/TX leads as a sanity check
6. changing "Boot sequence" delay to maximum and minimum values
7. Changing "Baud rate"

**What I found out:**
1. Voltages for RX/TX are 3.3V (no level shifter required)
2. I can connect to the printer via USB and Pronterface
3. I'm using the correct Baud Rate 115200
4. M117 command doesn't output anything on screen
5. Using pins TXD, RXD on the ESP bugs out the screen when both are connected. (meaning there is probably communication happening on those pins)

**PRINTER:**
Anycubic Kobra 2 
firmware v3.0.6
Marlin bugfix-2.1.x (Aug  2 2023 14:46:57) PROTOCOL_VERSION:1.0

Thread discussing UART port placement on a similar motherboard
https://klipper.discourse.group/t/anycubic-kobra-trigorilla-pro-a-v1-0-4-uart-to-pi-gpio-connection/15432

**ESP32:**
(ESP-WROOM-32) ESP32 Dev Board CH340 - USB-C
https://www.robotics.org.za/ESP32-DEV-CH340-C

Flashed using Visual Studio + Platform.io
build with esp32dev environment

**ESP3D:** branch 3.0
{
  "FWVersion": "3.0.0.a233",
  "FWTarget": "marlin",
  "FWTargetID": "20",
  "Setup": "Enabled",
  "SDConnection": "none",
  "SerialProtocol": "Raw",
  "Authentication": "Disabled",
  "WebCommunication": "Synchronous",
  "WebSocketIP": "192.168.0.14",
  "WebSocketPort": "81",
  "Hostname": "kobra2",
  "WiFiMode": "STA",
  "WebUpdate": "Enabled",
  "FlashFileSystem": "LittleFS",
  "HostPath": "/",
  "Time": "Manual",
  "Screen": "none",
  "HostTarget": "files",
  "HostUploadPath": "/",
  "HostDownloadPath": "/",
  "wsID": "0"
}

**ESP3d-WEBUI**: branch 3.0

luc-github, 2024-06-05:
what [ESP420] give as output ?

KristiyanHristov, 2024-06-05:
[ESP420]

Configuration:
chip id: 50220
CPU Freq: 240Mhz
CPU Temp: 46.7C
free mem: 188.69 KB
SDK: v4.4.4
flash size: 4.00 MB
size for update: 1.87 MB
FS type: LittleFS
FS usage: 104.00 KB/128.00 KB
baud: 115200
sleep mode: none
wifi: ON
hostname: kobra2
HTTP port: 80
Telnet port: 23
WebDav port: 8181
wifi mode: sta
mac: 08:D1:F9:E1:2C:C4
SSID: dd-wrt
signal: 100%
phy mode: 11n
channel: 1
ip mode: dhcp
ip: 192.168.0.14
gw: 192.168.0.1
msk: 255.255.255.0
DNS: 1.1.1.1
ap: OFF
mac: 08:D1:F9:E1:2C:C5
ntp: OFF
serial: ON (UART0)
authentication: OFF
notification: ON (none)
targetfw: marlin
FW ver: 3.0.0.a233
FW arch: ESP32
ok

KristiyanHristov, 2024-06-05:
yes USB is disconnected

luc-github, 2024-06-05:
what baud rate Marlin is set in your configuration.h of Marlin ?

KristiyanHristov, 2024-06-05:
I don't build my own printer firmware, but if this is the one they officially shipped
https://github.com/ANYCUBIC-3D/Kobra2/blob/Kobra2_V3.0.6/source/Marlin/Configuration.h

its #define BAUDRATE 115200

luc-github, 2024-06-05:
well if you cannot be sure it is hard to know if that is the root cause or not 

I suggest you to find a serial/usb adapter (something like https://www.amazon.com/HiLetgo-CP2102-Converter-Adapter-Downloader/dp/B00LODGRV8/ref=sr_1_1_ssp) and check if your Printer board wiring is correct and can communicate with your pc using serial terminal

luc-github, 2024-06-05:
also such device can be also used on esp32 to test the esp32 uart pins also

KristiyanHristov, 2024-06-05:
Ok. I'll source one locally. Will a six pin one do the job?

[image]

luc-github, 2024-06-05:
yes, you only need 3.3v gnd txt rxt pins, this will help you to debug the communications

KristiyanHristov, 2024-06-05:
Ok, I got the adapter and setup the terminal. What is the expected behavior? 
When connected to the printer I get some output:
[image]

luc-github, 2024-06-05:
nope that is not the Marlin output - do you have the same when you connect your printer and the serial terminal ?

KristiyanHristov, 2024-06-05:
No. When connected via USB I can send M503 and get the config.

[image]

luc-github, 2024-06-05:
well you found the root cause - the usb adapter do same as the CH340  so output should be same - it means the usb adapter is not connecting to rx/tx - to test use M105 or M114 or M115

KristiyanHristov, 2024-06-05:
Well, I have to do some poking around :)
You can close the ticket, I'll probably add a comment with my findings.

Thank you for your time and patience!

luc-github, 2024-06-05:
Good luck then ^_^

closed 2024-06-05 by luc-github

KristiyanHristov, 2024-06-05:
Everything is now running. I used pins 2 and 3 of the CH340.

[image]

luc-github, 2024-06-06:
cool feel free to add your board to the list with instruction and images:
https://esp3d.io/esp3d/v3.x/hardware/system_boards/index.html