luc-github/ESP3D #1035: ESP3D 3.0 git with Octoprint socket connection

SOliver78, 2024-07-08:
Hello,
I have a problem connecting ESP3D 3.0 git to Octoprint.
Octoprint 1.10.2 and Network-Printing Plugin (0.1.0) run with ESP3D 2.1.4 without any problems.
With ESP3D 3.0 git I can't get a connection with socket://192.168.110.23:8282, even though the port is open according to nmap.
Does anyone know the problem?

Why I want to use ESP3D 3.0 git? Because I want to use a WT32-ETH01 with LAN.

Thank you.
Olli

luc-github, 2024-07-08:
did you check the port is correct in your settings ?

SOliver78, 2024-07-09:
In ESP3D the following settings are set:
[image]
In Octoprint the connection is:
[image]
But after a few seconds, the status is offline:
[image]

The Octoprint server can reach the web socket port of the ESP3D:
[image]

Is it possible, to look into any connection logs on the EDP3D?

luc-github, 2024-07-09:
I am confused in 2.1 there is no websocket support for extra communication  but you mention it works, are you sure you are not using telnet in 2.1 ?

luc-github, 2024-07-09:
data port in 2.1 is the telnet  port in 3.0 , it is raw tcp actually 
so based on your setting use the port 23 for your Network-Printing Plugin 

using same port for different purpose between 2.1 and 3.0 is may be confusing, but it is not when you know what is for

luc-github, 2024-07-10:
I guess issue is now fixed - I close ticket then

closed 2024-07-10 by luc-github

SOliver78, 2024-07-10:
Hi, yes if you know that it is the telnet port, it works now ;)

Thanks for the hint.