luc-github/ESP3D #1037: [Question] Is this project still being supported/updated?

Reza-v, 2024-07-16:
I've had a mks tinybee for some time. terrible board, based on the esp32, it does not even support babystepping and mesh bed leveling, but the one part i absolutely loved about it was the esp3d. I'm trying to install it on an s2 board i have lying around and I've been at it for 3 days now and it's honestly just fighting back. i can't even seem to build it, always running into errors mostly related to the code not being adapted to newer libs and core updates, very frustrating. The "stable" v2.1.x seems to not even support s2 boards and the newer 3.0 which is still in alpha has a ton of problems that are beyond my skills to fix. On platformio it tries to download an outdated lib: " unknown package error: could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64' " perhaps due to the links being old, and on arduino ide it gives back the error "serial2 was not declared in this scope" and when you demolish serial2 from the face of the earth, it says "error cannot convert 'usbcdc*' to 'hardwareserial*' in assignment " Is there anything i'm missing? Or is there any bugfix version or something i can use?

luc-github, 2024-07-16:
mks tinybee do not use esp3d but esp3dlib with marlin
which does not support esp32-s2 neither esp32-s3

on other hand esp3d has S2 support and 3.0 does not have toon of issue  it is still in alpha becuse did not meet the number or feature planned to move to beta, both should compile out of the box.
please clarify your needs and your issues

you may also check http://esp3d.io
About
>Is this project still being supported/updated? 

Yes I am currently working on webui extension, project is huge and cover several repositories

Reza-v, 2024-07-17:
Processing esp32-s2 (platform: espressif32@5.1.0; board: esp32-s2-saola-1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
UnknownPackageError: Could not find the package with 'espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3' requirements for your system 'windows_amd64'

Reza-v, 2024-07-17:
i managed to fix it. now i am back to the serial2 error

```
esp3d/src/modules/serial/serial_service.cpp:38:61: error: 'Serial2' was not declared in this scope
 HardwareSerial * Serials[MAX_SERIAL] = {&Serial, &Serial1, &Serial2};
                                                             ^~~~~~~
esp3d/src/modules/serial/serial_service.cpp:38:61: note: suggested alternative: 'Serials'
 HardwareSerial * Serials[MAX_SERIAL] = {&Serial, &Serial1, &Serial2};
                                                             ^~~~~~~
                                                             Serials`
```

Reza-v, 2024-07-17:
i did it lol

luc-github, 2024-07-17:
S2 has only 2 serial , this was fixed a while ago , what version are you using ?

Reza-v, 2024-07-18:
update: i have successfully uploaded and tested the code on the esp32 s2 mini.

the fixes i've done:

- added the following line under [env:esp32-s2] in platformio.ini:

`platform_packages = espressif/toolchain-riscv32-esp`

this causes platformio to download the latest version of the mentioned toolchain and ignore the previous dependency that caused issues. could be that the link is dead or the version is no longer on the server.

- erased serial2 in the file serial_service.ccp line 38

weirdly this got the code working and uploaded.
new issue i'm facing is communicating with the 3d printer board. it's a atmega 2560 based mainboard so i've used a voltage divider to step down the 5v tx pin coming from the atmega to the esp32.
do i need another voltage divider on the rx pin as well? if i do well it's too late then...
i've tried the default pins (1,3) and tried setting my own pins in the pins.h file, no dice.

also as a side note, when i power the esp32 s2 and try to load up the ip address, it randomly loads up only sometimes. it did load up very reliably the first time but once i uploaded the webui firmware, it got very random, sometimes loads up and sometimes doesn't and shows me a black/white screen. could be fixed once i connect the serial to the board though.
another question i had is do i need to make any changes to my 3d printer firmware? my guess is not since i'm using the usb uart already used by marlin but doesn't hurt to ask.
thanks in advance.

luc-github, 2024-07-18:
I have just compiled the latest git for esp3d for S2 out of the box and I cannot duplicate any of your compilation issues:

[image]

So your changes are not necessary and not tested.
Because you did not answer my question, I cannot see what you are doing, so I cannot help you currently I have no idea of what your are doing

Reza-v, 2024-07-19:
In order to replicate the issue, try deleting your current set esp32 core in `C:\Users\<user>\. platformio` and toolchains and see if you can download all of them when attempting to build the code.

I am using v3.0 alpha3 downloaded from your esp3d repository
https://github.com/luc-github/ESP3D/releases/tag/v3.0.0-alpha3

luc-github, 2024-07-19:
that version is old and outdated please use the git directly

Reza-v, 2024-07-19:
i was under the impression that the non alpha 2.1 version does not support the s2. isn't the main page of the git v2.1?

luc-github, 2024-07-19:
the link for the development version (3.0) is in readme of 2.1
and https://esp3d.io list clearly what is supported on each version

Reza-v, 2024-07-20:
Hello luc
Thank you very much for providing me with the info. I had gotten my software from the releases tab so trying to install old versions was the main issue. Now the software side is successfully working with some tweaks to the configuration.h file.

Now i still haven't figured out how to connect the two boards. I've tried changing the pins, leaving the pins as -1 and tried pins 1,3. I am using a voltage divider to step down the voltage on the atmega's tx side but not the rx pin. What are the default rx tx pins on the esp32 if you leave them as -1? And does changing the rx/tx pins in the pins.h file work?

luc-github, 2024-07-21:
if it is a mega board people use the Serial0 of the board:
did you checked https://esp3d.io/esp3d/v3.x/hardware/system_boards/main_boards/ramps_mega/index.html
so no need any change in Marlin 
 I do not know what S2-Mini you use but https://esp3d.io/esp3d/v3.x/hardware/esp_boards/esp32-s2/wemos-s2-mini/index.html
 some are not supported yet because they only provide OTG Usb port which is not yet supported by ESP3D and do not expose the 
 Default Serial0 Pins: TX = 43, RX = 44; but Serial1 pins Rx = 4, Tx = 5 
So need to add  some documentation in esp3d.io that I have no time to do now

Reza-v, 2024-07-21:
Well that's a shame. Thanks again. Looking forward to seeing where this project goes

luc-github, 2024-07-21:
>Well that's a shame.

Why ?

Reza-v, 2024-07-21:
hello luc
upon further tries i still could not get it to work. i am stuck on this:
[image]
i tried hooking up the rx to pin 4 and tx to pin 5, with serial1 selected in the config file, tried default pins in pins.h and specifying pins 4 and 5, no dice.

my setup is 4 jumper wires, my 5v tx coming from the atmega is stepped down to 3.3v, confirmed with a multimeter but the tx pin coming from the esp32 is just directly connected to the atmega's rx.
here's my config and pins file, other files have not been modified:
[esp3d config.zip](https://github.com/user-attachments/files/16325483/esp3d.config.zip)
i must mention that i uncommented line 50 to be able to reliably open the ip address on a browser. with wifi tx power disabled, it would not load up the address, sometimes timing out and not load at all and sometimes taking very long only to give back an error saying:
`esp3d board does not answered properly 0`

Reza-v, 2024-07-21:
i also have a feature request. as the esp32 s2 platform natively supports hosting usb, you could perhaps hook it up to a usb camera and have a monitoring system using a regular usb webcam which would be super easy to hook up and use. i have seen people use an s2 to stream a usb camera feed over wifi so it is theoretically doable.

luc-github, 2024-07-21:
please open new issue with all informations requeted

luc-github, 2024-07-21:
> i also have a feature request. as the esp32 s2 platform natively supports hosting usb, you could perhaps hook it up to a usb camera and have a monitoring system using a regular usb webcam which would be super easy to hook up and use. i have seen people use an s2 to stream a usb camera feed over wifi so it is theoretically doable but i imagine it would take up lots of ram and flash space...

this is not planned sorry, it depend on what usb camera is plugged and may need extra drivers to read the stream

closed 2024-07-21 by luc-github